ngscopeclient/scopehal-apps #66: Check Minimum System Requirements on glscopeclient startup

antikerneldev, 2020-02-24:
Verify the GPU can actually support the required feature set (Like OpenGL 4.3 for compute shaders) and throw a warning/error otherwise.

antikerneldev, 2020-02-25:
Committed incomplete test to a new branch opengl_version_check (bea1a89846faf7318f5346a08c71a09b8b31221f).

Issues:

```C
/* FIXME: This code does not currently function correctly.
 * Issues:
 *
 * - The version returned is not the highest OpenGL version the system supports.
 *   This is due to how GTK is creating the OpenGL context, and how OpenGL deals with backwards compatibility.
 *   We should probably have GTK create the OpenGL context, requesting version 4.3 as the minimum required version.
 *
 *   Reference Links:
 *   - https://stackoverflow.com/questions/47151394/how-to-tell-which-version-of-opengl-my-graphics-card-supports-on-linux/47166829#47166829
 *   - https://stackoverflow.com/questions/46510889/how-can-i-know-which-opengl-version-is-supported-by-my-system/46511741#46511741
 *   - https://cgit.freedesktop.org/mesa/demos/tree/src/xdemos/glxinfo.c
 *
 */
```

azonenberg, 2020-02-25:
WaveformArea now requests 4.3 core profile during startup. Can you re-test your branch with this change and see if it properly detects what we got?

azonenberg, 2020-09-05:
Closing as fixed in 926ada64d1b9b2be380bb3f92ae2b89c28f833e9

closed 2020-09-05 by azonenberg