ngscopeclient/scopehal-apps #106: Option to decouple changing the Y axis in glscopeclient from changing the Y axis settings on the scope

tomverbeure, 2020-05-26:
Right now, when you change the Y axis in glscopeclient, these changes get transmitted to the oscilloscope.

This is not always desired: you may want to zoom in on an acquired signal without changing the settings of the scope.

There should be an option to do that.

macaba, 2024-09-08:
(Note this is written from my point of view, it's not an assumed point of view for the majority)

Summary of a discussion elsewhere:

- If zooming horizontally, you're inspecting the waveform at your leisure. The sample depth is a control on the timebase dialog. This feels natural.
- If zooming vertically, you're not inspecting the waveform at your leisure, you're updating the hardware frontend (with all the pitfalls that could have with multi-trace displays). This overwrites the controls on the channel dialog. This feels like it's working against me. There's additional concerns when working with multiple channels on the same plot that have very different vertical settings.

The vertical controls could be decoupled from the display control.
Both approaches have pros/cons for different users so the current thought is to make this a switchable feature. Perhaps with the "disconnected" being true by default for new users.

Added benefit: having hardware control decoupled from display control becomes even more relevant for 12-bit scopes where it might be easier to capture the full height of a waveform, then zoom into a feature of interest.

Following on from that:

- Could there be a compact way to display the channel offset/range? It's not hopeless at the moment - it's possible to dock the channel configuration dialogs though it requires many clicks and isn't compact. Would it be better to have a new user default that displays these crucial settings somewhere?

jwise, 2024-09-30:
Some discussion of this in Discord when I tripped over this today.  I agree that X and Y behaving differently is definitely an 'at least one of them is wrong' situation.

One idea I am toying with is that a waveform view should, metaphorically, represent a capture of a state of the system, and might have indications for 'these were the parameters this capture happened with', and also indications for 'these parameters are what will apply to the next capture'.  Consider, for instance:

* A dashed line and flags at right indicating a channel's minimum and maximum Y-axis (middle-clicking for 'full scale' will return to this as the extent of the viewing area)
* A tightly dotted line and flags at right indicating the Y-axis that will be programmed to the instrument in the *next* capture.  These can be dragged with the left mouse button to adjust them to set up for the next capture.
* Zooming on the Y axis only modifies the view, it does not modify the parameters of the instrument.
* Mis-clicks happen.  There should be some UI to indicate 'the next capture does not have the same parameters as the current capture; click here to reset the parameters of the next capture to be identical to this one'.
* What happens if you zoom in too close on the Y axis, such that you are meaningfully running out of bits?  The tool will provide a ⚠️ indicator in the Y axis to say 'there are not many samples, and there is a lot of interpolation happening here'.