ngscopeclient/scopehal-apps #133: Commission toolbar icons for "auto trigger" and "force trigger"

azonenberg, 2020-07-09:
Need to come up with a creative brief and $100 to pay the artist.

mubes, 2021-06-06:
For auto trigger perhaps an arrow with a circular arrow around it (i.e. a 'repeat' indication). For Force trigger perhaps a lightning bolt or something? For me the simpler the better...e.g. some of the examples at https://www.iconfinder.com/search/?q=trigger

jwise, 2024-10-02:
I was also thinking that the common metaphor is the 'play' icon, but in 24x24 there is not much space for a circle-around 'repeat'.  Originally I was also kind of thinking about this of the 'repeat' arrow going from the right side of the play button to the left to indicate 'repeat', but my current thought is:

* Single capture, from trigger: play button with "T" inside it
* Continuous capture, from trigger: a "stack" of play buttons, with "T" in the top play button
* Continuous capture, autotrigger: a "stack" of play buttons with "A" in the top play button
* Single capture, immediate: a single play button with an "!" inside it (note: the semantics of this are also quite nonintuitive, currently 'single capture immediate' when connected to my Rigol means 'capture, if a capture is pending', while I would expect it to mean 'whether or not there is a capture pending, capture as soon as you can')
* Stop capture: a stop button (or play button with an X through it?)

Other thoughts while I'm in there:
* I originally like the clock metaphor for history, but it's really hard to see.  What about three play buttons in a horizontal line, with two of them dotted or something?  A ⏪ button is also an option but that sort of connotes a verb rather than browsing some nouns, at least for me.
* Flush state is sort of challenging because it introduces an entirely new icon metaphor: "instrument" and "host".  Where else do we need to display the concept of 'instrument' or 'host' in icons?  (Aside: I really really want to use, like, ⚓ as a metaphor for "instrument".  But it is hard to fit in that space and also probably not as clear as it should be, despite that it would be extremely funny.)
* Does fullscreen mode belong on the toolbar at all?  Is that not an OS-specific thing?  Like on OS X you just click the 'fullscreen' button and you get it.