ngscopeclient/scopehal-apps #137: Opening ethernet-example.session livelocks (?) glscopeclient

whitequark, 2020-07-10:
Doesn't render anything, stuck with backtraces like this one:
```
#0  0x00007ffff7fd2abe in clock_gettime ()
#1  0x00007ffff618e2e7 in __GI___clock_gettime (clock_id=0, tp=0x7fffffffc530) at ../sysdeps/unix/sysv/linux/clock_gettime.c:33
#2  0x00005555555fd20e in GetTime() () at /home/whitequark/Projects/scopehal-cmake/src/glscopeclient/main.cpp:303
#3  0x00005555555b8bce in OscilloscopeWindow::OnAllWaveformsUpdated() (this=0x5555557574a0)
    at /home/whitequark/Projects/scopehal-cmake/src/glscopeclient/OscilloscopeWindow.cpp:2021
#4  0x00005555555bbad8 in OscilloscopeWindow::PollScopes() (this=0x5555557574a0)
    at /home/whitequark/Projects/scopehal-cmake/src/glscopeclient/OscilloscopeWindow.cpp:1965
#5  0x00005555555bbc25 in OscilloscopeWindow::OnTimer(int) (this=0x5555557574a0)
    at /home/whitequark/Projects/scopehal-cmake/src/glscopeclient/OscilloscopeWindow.cpp:403
#6  0x00007ffff7653b12 in  () at /usr/lib/x86_64-linux-gnu/libglibmm-2.4.so.1
#7  0x00007ffff691a863 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#8  0x00007ffff6919dd8 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#9  0x00007ffff691a1c8 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#10 0x00007ffff691a4c2 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#11 0x00007ffff70ab583 in gtk_dialog_run () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#12 0x00005555555a3d22 in OscilloscopeWindow::DoFileOpen(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, bool, bool)
    (this=this@entry=0x5555557574a0, filename="ethernet-example/ethernet-example.scopesession", loadLayout=loadLayout@entry=true, loadWaveform=<optimized out>, reconnect=reconnect@entry=false) at /home/whitequark/Projects/scopehal-cmake/src/glscopeclient/OscilloscopeWindow.cpp:592
#13 0x00005555555d413c in ScopeApp::run(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, bool, bool, bool)
    (this=this@entry=0x555555633a60, fileToLoad="ethernet-example/ethernet-example.scopesession", reconnect=reconnect@entry=false, nodata=nodata@entry=false, retrigger=retrigger@entry=false, nodigital=nodigital@entry=false) at /usr/include/c++/8/bits/char_traits.h:287
#14 0x00005555555a9ab2 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /usr/include/c++/8/bits/basic_string.h:936
```

azonenberg, 2020-10-06:
I think this was fixed by the patch for #214. If you still have problems reopen the ticket.

closed 2020-10-06 by azonenberg

whitequark, 2020-10-06:
> If you still have problems reopen the ticket.

Not important for this case, but just so you're aware: I can't reopen this issue since I'm not a collaborator. The UI looks like this:
[image]