ngscopeclient/scopehal-apps #144: Add ffts as a submodule

whitequark, 2020-07-13:
This is annoying to do manually (I distinctly remember having to screw with it to get it to work at least twice for scopehal), and might fix #139.

karlp, 2022-09-20:
Is there any reason not to just use fftw? it's well packaged and widely available, ffts hasn't had any commits in 6 years and it's website is offline?

azonenberg, 2022-09-20:
Yes, the license incompatibility is the primary reason we didn't use FFTW. There is an extreme lack of permissively licensed FFT libraries and it was the best option available at the time.

We are actually in the process of moving away from FFTS in favor of vkFFT (https://github.com/glscopeclient/scopehal/issues/660) so this ticket should have been closed as wontfix.

Closing now.

closed 2022-09-20 by azonenberg