ngscopeclient/scopehal-apps #180: Look into static analysis options

azonenberg, 2020-09-21:


bvernoux, 2020-10-01:
A good start will be to always check the source code (especially new source code) with CppCheck 2.1 (or more) http://cppcheck.sourceforge.net/

tarunik, 2020-10-02:
As long as the code compiles with clang (which may be worthwhile on its own for diagnostics/compiler-portability reasons), we should give the [clang-analyzer](https://clang-analyzer.llvm.org/) a whirl as well.

azonenberg, 2020-10-02:
Let's start with cppcheck. @tarunik can you work on that?

My thought is to detect cppcheck and (if present) create a new build target "analysis" which is built by default as part of "make all", which runs cppcheck and displays all detected error messages.

tarunik, 2020-10-02:
Yeah -- it looks like CMake has built-in support for at least cppcheck and clang-tidy (albeit not the full clang-analyzer?): https://blog.kitware.com/static-checks-with-cmake-cdash-iwyu-clang-tidy-lwyu-cpplint-and-cppcheck/

azonenberg, 2020-10-02:
Oh even better. Add cppcheck integration and send a PR when you're ready? We'll close this ticket when that's done, then think about other static analyzers in the future if we see a need for it.

tarunik, 2020-10-09:
Did you want static analysis mode to require all the static analyzers to be present? (so far, it'll be cppcheck and clang-analyzer, clang-tidy will likely go on the list as well)

azonenberg, 2020-10-09:
Ideally it should detect each individually and enable as many as it find and knows how to use.

azonenberg, 2020-10-12:
Merged, tested, and fixed a bunch of findings.

On my machine, cppcheck fails silently (no errors displayed but nonzero exit code) on VICPSocketTransport.cpp.