ngscopeclient/scopehal-apps #209: glscopeclient Fullscreen issue with dialog box

bvernoux, 2020-10-01:
How to reproduced the issue (tested on Windows 7 Pro 64bits):
- Launch glscopeclient and choose Fullscreen option 
- if you go to Help=>About... or File=>Open... or any Dialog box
- The dialog box is displayed in background and so it is not visible and not usable..

This issue has been reproduced with Windows build but I suspect it can be reproduced also on Linux to be confirmed

bvernoux, 2020-10-01:
To be checked if this hint https://stackoverflow.com/questions/11865150/let-a-window-display-above-a-fullscreen-window-in-gtk is applicable and fix the issue

azonenberg, 2020-10-09:
Fullscreen works fine for me on Debian with XFWM4 and the current glscopeclient revision, but was a source of lots of problems with popup dialogs in the past. I'm not surprised to hear it fails on other platforms.