ngscopeclient/scopehal-apps #210: Example scope session files

bluecmd, 2020-10-05:
Hi,

I was checking out glscopeclient but I do not have any of the supported scopes.

Would you consider uploading some .scopesession files that can be used to play with the features?
I found an `ethernet-example.scopesession` linked from the `##openfpga` IRC channel, but when loading that I only get a black screen. I am not sure if that is my setup or if it is related to #137 

Anyway, having a few example sessions would probably be nice for any future training or tutorials.

azonenberg, 2020-10-05:
I have a bunch of demo files hosted internally, but many of them are fairly large because they're either very long, acquired at high sample rates, or just have a ton of waveforms.

Try https://www.antikernel.net/temp/10gbe.tar.gz and see if you have problems loading it. If you do, please open a separate issue for that. Include a screenshot and the console output from running glscopeclient with the --debug argument.

Leaving this issue open until we get a repo/download location with official demo waveforms set up (trimmed down to minimal examples without excessive sample rate/depth to keep them small),

azonenberg, 2022-02-25:
https://github.com/azonenberg/scopehal-testdata has existed for quite a while, somehow I never closed this issue. Fixed.

closed 2022-02-25 by azonenberg