ngscopeclient/scopehal-apps #211: Black screen when starting glscopeclient

bluecmd, 2020-10-05:
Hi,

I am using the example 10gbe.tar.gz file provided in #210. I am at commit ce513eb825dee588686d1cdeed95084f33d7af35.
I am running using Ubuntu 20.04 under WSL (not WSL2). I am using the [X410](https://x410.dev/) X server.

I start glscopeclient like this: `src/glscopeclient/glscopeclient  ../../10gbe/64b66b.scopesession --debug`

The output is:
```
Warning: glscopeclient works best with the OMP_WAIT_POLICY environment variable 
set to PASSIVE
Detecting CPU features...
    * AVX2
``` 

Here is a screenshot:
[image]

azonenberg, 2020-10-05:
First, try rerunning glscopeclient with OMP_WAIT_POLICY=PASSIVE and see if that helps. We've had issues with 100%-CPU hangs or slow performance if that's not set.

But I have no idea what kind of graphics support is present under WSL as far as OpenGL goes. We've never tested this. The fact that you don't see any output at all (even menus/toolbars) is concerning, as is the fact that there's no debug logging listing GL_VENDOR etc. You're not even reaching the point of having an OpenGL context to work with.

If you aren't using native Linux, you should probably try following the MinGW/msys2 Windows build instructions in the manual to produce a native Windows binary.

azonenberg, 2020-10-05:
Root cause identified. Closing as duplicate of #214.

closed 2020-10-05 by azonenberg