ngscopeclient/scopehal-apps #212: glscopeclient requires GL 4.3 even though we don't need most 4.3 features

azonenberg, 2020-10-05:
We can probably get away with GL 3.x as long as we have GL_ARB_compute_shader and GL_ARB_gpu_shader_int64. This might allow running in VMWare?

azonenberg, 2020-10-05:
Minimum requirement is now 4.2 plus a bunch of extensions. It looks like glBindImageTexture is the main thing we need from 4.2, can we go even further down with extension checks or finding another way to do this?

nshcat, 2020-10-05:
The question is whether the set of GPUs that happen to both support the extensions and at the same time not OpenGL 4.2 is sufficiently large to make it worth the effort.

azonenberg, 2020-10-05:
The immediate use case is WSL. Here's a glxinfo dump:

https://gist.github.com/bluecmd/e2c5bd22a5fa70a1e32541a2e5540c49

It reports OpenGL 3.3 core profile plus, AFAIK, all of the extensions we need.

azonenberg, 2020-10-05:
As far as VMware goes, it looks like Workstation 15 will not work (it's missing compute shaders among other extensions), but if we can get down to 4.1 plus extensions there's a good chance that Workstation 16 will work. I might upgrade my Workstation license to 16 soon since it seems like the list of GL extensions it supports is hard to find.

Haven't looked at other hypervisors yet.

xzcvczx, 2021-04-24:
Intel Ivy Bridge also supports compute shaders with recent versions of mesa on Linux. however I needed to use 

```MESA_GL_VERSION_OVERRIDE=4.3 MESA_GLSL_VERSION_OVERRIDE=430```

 in order for it not to error out

azonenberg, 2022-09-10:
Closing as wontfix now that the long term plan is to transition everything to Vulkan.

closed 2022-09-10 by azonenberg