ngscopeclient/scopehal-apps #215: Window title of protocol analyzer windows not shown or imcomplete

electroniceel, 2020-10-05:
On some linux window managers (or desktop environments, not sure) the window title of protocol analyzer windows is not shown at all, on some it is missing things like the "X" to close or the title or correct colors/style.

I tested these desktop environments with their default window managers:
KDE: incomplete
Xfce: correctly shown, "X" to close works
awesome: complete title block missing
Cinnamon: complete title block missing

All tests were done with Fedora 32, x86_64, X.org, AMD GPU.

This problem was also seen by user bluecmd[m]1 when running glscopeclient in WSL: https://www.youtube.com/watch?v=E6XVPqGP0gA

Screenshot from KDE:
[image]

nshcat, 2020-10-05:
I can reproduce and confirm this, on my Linux laptop with awesomeWM, the title block is missing.

azonenberg, 2020-10-09:
Please test and see if 361221f fixes it.

nshcat, 2020-10-09:
The linked commit indeed fixes the issue for AwesomeWM.

azonenberg, 2020-10-09:
Seems like everyone affected has confirmed the fix. Closing.

closed 2020-10-09 by azonenberg

electroniceel, 2020-10-09:
Fixed for me on KDE too