ngscopeclient/scopehal-apps #216: Wave source hierarchy browser

bluecmd, 2020-10-06:
In order to support sources with a lot of channels, it is inevitable that glscopeclient will have to offer some sort of interface to select and add a lot of channels at once. Right now this is done by the "Add" menu, which works pretty well for a few channels commonly used on analog scopes.

The most extreme examples I can think of are SPICE simulations which can easily have thousands of nets in the results, 
I made an example SPICE waveform here: [lots-of-inverters.tar.zst](http://storage.googleapis.com/bluecmd/lots-of-inverters.tar.zst). It contains a 300MB SPICE simulation of some hundred inverters in a 2-level hierarchy.

Some inspiration from GTKwave here:
[image]

Other simulation tools usually have something similar, a type of "filesystem" folder/trace structure that allows navigation. Some viewers have a flat hierarchy but allowing you to filter.

For VCD and FST the folder-hierarchy maps really well as they are hierarchical anyway. For SPICE data they are more implicit, like this:

```
v(x16.net30)    voltage
v(x16.net31)    voltage
v(net16)        voltage
v(sky130_fd_pr__esd_nfet_05v0_nvt.pm3)  voltage
v(m.x16.xm60.msky130_fd_pr__nfet_01v8#dbody)    voltage
v(m.x16.xm60.msky130_fd_pr__nfet_01v8#body)     voltage
i(v.x3.vsense-vcc)      current
i(v.x2.vsense-in)       current
i(v.x2.vsense-vcc)      current
i(v.x1.vsense-in)       current
i(v.x1.vsense-vcc)      current
i(v1.8) current
i(v.100mhz)     current
```

bluecmd, 2020-10-06:
For reference, this is what gaw does for SPICE right now:
[image]

That's a long list of unfilterable SPICE data names. As a user that is pretty annoying and not so friendly, so let's avoid that :-)

bluecmd, 2020-10-06:
After speaking with @azonenberg I will post some commercial examples as well.

Here is Cadence's Visualization tool showing a quite busy setup with SPICE data:
[image]
[image]

Here is Synopsys WaveView:
[image]
[image]

azonenberg, 2020-10-06:
Anybody have a screenshot of VCS for reference? The only ones I have are full of customer data and I can't share.

bluecmd, 2020-10-06:
Maybe these are from VCS? https://github.com/pulp-platform/fpnew/issues/24#issuecomment-613736398

commit 6e088de4 (2024-08-26): Can now drag streams from the browser into the filter graph editor. Fixes #216.
closed 2024-08-26 by azonenberg