ngscopeclient/scopehal-apps #239: Continuous integration setup

azonenberg, 2020-10-20:
We should have nightly Windows and Linux builds logging to a CI server.

azonenberg, 2020-10-21:
@bvernoux @nshcat can one of you investigate these Windows build failures?

https://github.com/kench/scopehal-apps/actions/runs/319193878

azonenberg, 2020-10-22:
@kench looks like your OOM issue is the result of building with "make -j" and no cap on the number of jobs. If you have 192GB of RAM on your workstation like I do this is fine, but GitHub's build system clearly has less :)

kench, 2020-10-22:
@azonenberg Can fix. While we're on the topic of **make -j**, can you update the glscopeclient manual?

It's going to be frustrating for new users when their Windows builds fail because the instructions assume that they have 192GB of RAM installed.

azonenberg, 2020-10-22:
Updated in https://github.com/azonenberg/scopehal-docs/commit/549cf1efd6ec566f0733b7a1b06a218012212ffd

closed 2020-10-23 by azonenberg