ngscopeclient/scopehal-apps #249: Rigol MSO5074 crashes when we try to do stuff with glscopeclient

kench, 2020-10-24:
I need to include packet captures, but both the scope and glscopehal become unresponsive.

azonenberg, 2020-10-24:
Possibly related to #146.

To avoid overlap, this ticket specifically for the *scope firmware* hanging, not glscopeclient.

electroniceel, 2020-10-24:
Degi was able to reproduce the crash with this short code snippet: http://paste.debian.net/1165913/

(Posted on IRC here: https://freenode.irclog.whitequark.org/scopehal/2020-10-04#28061962 )

bvernoux, 2020-10-25:
I cannot reproduce that issue with my Rigol MSO5000 (HW 01.01.00) and latest firmware 01.03.00.01 
Anyway so far on windows glscopeclient is unusable with something less than 1 frame per second...

azonenberg, 2020-11-05:
Believed fixed with latest Rigol firmware.

closed 2020-11-05 by azonenberg