ngscopeclient/scopehal-apps #264: No .cmake file for Catch2

Cushychicken, 2020-10-30:
Got the following error while following install instructions in README.md: 

```-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found Git: /usr/bin/git (found version "2.25.1") 
CMake Error at CMakeLists.txt:61 (find_package):
  By not providing "FindCatch2.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Catch2", but
  CMake did not find one.

  Could not find a package configuration file provided by "Catch2" with any
  of the following names:

    Catch2Config.cmake
    catch2-config.cmake

  Add the installation prefix of "Catch2" to CMAKE_PREFIX_PATH or set
  "Catch2_DIR" to a directory containing one of the above files.  If "Catch2"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/nreilly/projects/scopehal-apps/build/CMakeFiles/CMakeOutput.log".
```
Guessing this is some install dependency that hasn't been added yet? I'm no expert in CMake or unit testing but a quick search of the issue backlog turned up #241 which references Catch2 being added as a submodule for unit tests. 

Happy to provide more info. Eager to try this sucker out. :D

nshcat, 2020-10-30:
Okay, yes, this seems to be an issue. In the meantime, you can disable the need for catch2 by first completely emptying out your build directory (to clear the cache) and doing:

`cmake -DBUILD_TESTING=Off ./..`

azonenberg, 2020-10-30:
I don't think we've fully debugged unit testing on Windows yet. Disabling testing with the command mentioned above is probably the best option for now. If you're not a developer writing new code, there's no point in having tests enabled so they should probably be off by default in new build trees moving forward?

Cushychicken, 2020-10-30:
> To help with the issue - do you have catch2 installed on your system using your package manager?

@nshcat unless it's a default I don't know about on Ubuntu 20.04, I do not have catch2 installed. (Didn't know what it was til I searched thru the open issues. XD ) 

@azonenberg - trying this on Ubuntu 20.04 if that helps.

azonenberg, 2020-10-30:
@Cushychicken catch2 is not available in distro packaging until 20.10. For 20.04 your options are either to build it yourself from source, or to just disable testing using -DBUILD_TESTING=Off.

Did you read the "getting started" section in the manual I linked you? This was documented pretty clearly there on page 12.

Cushychicken, 2020-10-30:
@azonenberg - nope, I did not. Bracing for the "RTFM" here, but I started from the README.md page. Can I help out with a PR to make the README installation instructions match the manual? 

@nshcat Cleared the `build` directory (`rm -rf ./build`), ran the suggested command, and it now appears to be building happily. :smiley:

azonenberg, 2020-10-30:
This is still valuable feedback to us. We don't have a large team writing documentation so stuff often gets overlooked. Problems new users run into are important because they can scare off future users.

PRs are very welcome. That said, now that you know, I suggest you read the manual :) It's definitely incomplete, but it should give you a good feel for the basics of how to use the software.

Cushychicken, 2020-10-30:
> I suggest you read the manual :) It's definitely incomplete, but it should give you a good feel for the basics of how to use the software.

Of course! It just finished compiling, so I'll be moseying on over to the Rigol section of the manual as soon as I finish that PR. 

Thanks for the help, y'all. Excited to see the fruits of your labor. :smile:

closed 2020-10-30 by azonenberg