ngscopeclient/scopehal-apps #274: ASAN reported new/delete size mismatch (sliced object?)

tarunik, 2020-11-01:
I started up glscopeclient as shown below with my Rigol DS1054Z powered on, and passed it the default name, rigol for the driver, usbtmc for the transport, and /dev/usbtmc1 as the device (which is the correct device), and I get the following crash after tabbing to the OK button and hitting enter:

> $ OMP_WAIT_POLICY=PASSIVE ./glscopeclient 
=================================================================
==1655737==ERROR: AddressSanitizer: new-delete-type-mismatch on 0x60800004d520 in thread T0:
  object passed to delete has wrong type:
  size of the allocated type:   88 bytes;
  size of the deallocated type: 48 bytes.
    #0 0x7f502c4f2467 in operator delete(void*, unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.6+0xac467)
    #1 0x55d737305e9b in sigc::slot0<void>::~slot0() /usr/include/sigc++-2.0/sigc++/functors/slot.h:515
    #2 0x55d737305eb7 in sigc::slot<void, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil>::~slot() /usr/include/sigc++-2.0/sigc++/functors/slot.h:1761
    #3 0x55d73734f00a in OscilloscopeWindow::CreateWidgets(bool, bool) ~/scopehal/src/glscopeclient/OscilloscopeWindow.cpp:286
    #4 0x55d737348148 in OscilloscopeWindow::OscilloscopeWindow(std::vector<Oscilloscope*, std::allocator<Oscilloscope*> > const&, bool, bool) ~/scopehal/src/glscopeclient/OscilloscopeWindow.cpp:82
    #5 0x55d737430c05 in ScopeApp::run(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, bool, bool, bool, bool) ~/scopehal/src/glscopeclient/ScopeApp.cpp:50
    #6 0x55d7374e572b in main ~/scopehal/src/glscopeclient/main.cpp:296
    #7 0x7f502a89dcc9 in __libc_start_main ../csu/libc-start.c:308
    #8 0x55d7372ed159 in _start (~/scopehal/build/src/glscopeclient/glscopeclient+0x1f1159)

0x60800004d520 is located 0 bytes inside of 88-byte region [0x60800004d520,0x60800004d578)
allocated by thread T0 here:
    #0 0x7f502c4f1647 in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.6+0xab647)
    #1 0x55d737399bf0 in sigc::slot0<void>::slot0<sigc::bound_mem_functor0<void, OscilloscopeWindow> >(sigc::bound_mem_functor0<void, OscilloscopeWindow> const&) /usr/include/sigc++-2.0/sigc++/functors/slot.h:546
    #2 0x55d73738948a in sigc::slot<void, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil>::slot<sigc::bound_mem_functor0<void, OscilloscopeWindow> >(sigc::bound_mem_functor0<void, OscilloscopeWindow> const&) (~/scopehal/build/src/glscopeclient/glscopeclient+0x28d48a)
    #3 0x55d73734efd8 in OscilloscopeWindow::CreateWidgets(bool, bool) ~/scopehal/src/glscopeclient/OscilloscopeWindow.cpp:286
    #4 0x55d737348148 in OscilloscopeWindow::OscilloscopeWindow(std::vector<Oscilloscope*, std::allocator<Oscilloscope*> > const&, bool, bool) ~/scopehal/src/glscopeclient/OscilloscopeWindow.cpp:82
    #5 0x55d737430c05 in ScopeApp::run(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, bool, bool, bool, bool) ~/scopehal/src/glscopeclient/ScopeApp.cpp:50
    #6 0x55d7374e572b in main ~/scopehal/src/glscopeclient/main.cpp:296
    #7 0x7f502a89dcc9 in __libc_start_main ../csu/libc-start.c:308

SUMMARY: AddressSanitizer: new-delete-type-mismatch (/usr/lib/x86_64-linux-gnu/libasan.so.6+0xac467) in operator delete(void*, unsigned long)
==1655737==HINT: if you don't care about these errors you may set ASAN_OPTIONS=new_delete_type_mismatch=0
==1655737==ABORTING

azonenberg, 2020-11-01:
This has happened to me since I started using asan. It appears to be something deep in the guts of sigc++ and not anything we can fix in our code.

If you can find a proper fix/workaround then great, otherwise just use that asan env var to suppress it while looking for real bugs.

tarunik, 2020-11-01:
It seems that we're hitting https://github.com/libsigcplusplus/libsigcplusplus/issues/10 which is WONTFIX for sigc++ 2.x (we'd need to update to sigc++ 3.x to fix it, as the fix requires breaking ABI compat)

azonenberg, 2020-11-01:
So I guess we can close this issue then?

- Nothing we can do about it
- Will go away on its own when we upgrade to sigc++ 3.x in the future
- Harmless

Maybe put a note in the manual about this somewhere so other devs know, then just set that asan variable for debug builds until then?

tarunik, 2020-11-01:
I'll have to look and see if there's a finer-grained way to suppress this specific error...

tarunik, 2020-11-02:
It looks like we'll have to use a compile-time blacklist for this error since it's coming from code in a header file...

tarunik, 2020-11-05:
...and it turns out -fsanitize-blacklist is a clangism.  Using ASAN_OPTIONS=new_delete_type_mismatch=false works for now, but isn't particularly desirable since that shuts off that check for our code as well.

azonenberg, 2020-11-05:
Closing as unfixable for now, will update to sigc++3 once all supported platforms have it available.

closed 2020-11-05 by azonenberg