ngscopeclient/scopehal-apps #277: glscopeclient fails to run on Fedora under Wayland due to GLEW init failure

azonenberg, 2020-11-04:
glewInit returns GLEW_ERROR_NO_GLX_DISPLAY. Seems like something is looking for GLX when we're using Wayland and things bork.

xzcvczx, 2021-04-24:
GDK_BACKEND="x11" seems to work if you have xwayland

azonenberg, 2021-04-28:
Do we have a working flow for getting glscopeclient to run under Wayland, perhaps using xwayland? If so, please submit a PR to scopehal-docs with instructions on how to get this working.

Keeping this issue open and tagged for the v1.0 release because I want better native Wayland compatibility long term.

dannas, 2021-10-20:
Over at the GLEW repo, https://github.com/nigels-com/glew/issues/172, the supertux folks reports that GLEW seems to be usable even if `glewInit` returns GLEW_ERROR_NO_GLX_DISPLAY.

I can start glscopeclient and display the waveforms for the demo device, if I make this modification:

```
diff --git a/src/glscopeclient/WaveformArea.cpp b/src/glscopeclient/WaveformArea.cpp
index e68601b..d8dfc09 100644
--- a/src/glscopeclient/WaveformArea.cpp
+++ b/src/glscopeclient/WaveformArea.cpp
@@ -504,7 +504,7 @@ void WaveformArea::on_realize()
 
                //Initialize GLEW
                GLenum glewResult = glewInit();
-               if (glewResult != GLEW_OK)
+        if (glewResult != GLEW_OK && glewResult != 4)
                {
                        string err =
                                "glscopeclient was unable to initialize GLEW and cannot continue.\n"
```
The Glew maintainer said that he would expose  `glewContextInit` in the public API which would allow for both EGL and GLX to be used (if I understood him correctly). But that was back in 2018 and no such change has been committed.

I guess we can use the glew-patch that supertux uses: https://github.com/supertuxkart/stk-code/commit/961ac4d79eda230722d270bbf46f1545b324d0c8, but then we'd have to package our own version of glew.

Or we can check for  $XDG_SESSION_TYPE == wayland and in that case ignore GLEW_ERROR_NO_GLX_DISPLAY? I'll google a bit to check what is the most reliable way of detecting if a program is running under wayland.

PR #391 (2021-10-30): WaveformArea: Ignore GLX errors for Wayland
With this change, I can start glscopeclient on my Ubuntu 21.04 laptop. I can open the demo device and display the waveforms.

It relies on the undocumented behavior that libglew is properly initialized, even though it has reported GLEW_ERROR_NO_GLX_DISPLAY.

Here's the definition glewInit for the 2.0.0 release dating back to 2016. It has stayed the same AFAICT in 2.1.0 and 2.2.0: https://github.com/nigels-com/glew/blob/glew-2.0.0/auto/src/glew_init_tail.c#L35 

I have libglew-dev 2.1.0-4 installed.

It's a hack that hopefully can be removed once GLEW [... cut]
closed 2021-10-30 by azonenberg