ngscopeclient/scopehal-apps #284: Agilent MSOX2024A - Performance issues and crash

attie, 2020-11-20:
I've been having some performance issues in glscopeclient later this afternoon (it was only drawing a few of the captured waveforms, and was often completely unresponsive). Restarting glscopeclient didn't help.

Both glscopeclient and my scope have just crashed - the InfiniiVision software died, it threw up a plain text warning on a black background, and mentioned that it was checking the filesystem integrity (I was too slow with a camera and didn't manage read the full message), before rebooting itself.

On the subsequent startup, I was greeted with the following message.

[image]

Now, starting glscopeclient again, the performance issues appear to have disappeared. I also note that the waveform stats and trigger voltage are back in the status bar, along with the trigger arrow (they were completely missing while I was having the performance issues).

[image]

[image]

The output of glscopeclient on the command line when it died is as follows:

```
Warning: Unknown trigger type "GLIT"
Warning: Unknown trigger type "GLIT"
Warning: Unknown trigger type "GLIT"
Warning: Unknown trigger type "GLIT"
Warning: Unknown trigger type "GLIT"
Warning: Unknown trigger type "GLIT"
Warning: Unknown trigger type "GLIT"
    Warning: Socket read failed (errno=0)
    Warning: Socket read failed (errno=0)
    Warning: Socket read failed (errno=0)
    Warning: Socket read failed (errno=0)
Segmentation fault
```

It's worth mentioning that the `Warning: Unknown trigger type "GLIT"` messages are no longer being presented, where previously they were presented on every single capture.

I was running the build I made from ea5b3db the other day.

---

I don't know how best to progress this... I wonder if part of the answer is related to a memory leak or similar on the scope itself, as I've had things running for many hours today.

Please let me know if you'd like any more information.

miek, 2020-11-20:
The driver doesn't yet handle glitch triggers so I think that would cause glscopeclient/scopehal to poll for the trigger config, throw that `Warning: Unknown trigger type "GLIT"` message and fail to cache the result - leading to it polling constantly and ruining performance.
The reboot would've reset to edge trigger and I think that's why the warnings went away and performance improved.

attie, 2020-11-20:
Very interesting, thanks for the info @miek.

I don't think I was using a "_glitch trigger_", though if that's a class, I may well have been using one that is covered by that. I'll investigate next time I have things running.

miek, 2020-11-21:
Were you using the Pulse Width trigger on the scope instead? For some reason it's called "glitch" in the SCPI commands, but Pulse Width throughout the scope UI (on my MSO6034A, anyway).

azonenberg, 2021-04-23:
Any updates on this? Is this still a problem or is it fixed in recent builds?

attie, 2021-04-25:
Hi Both,

Very sorry - I thought I'd responded to @miek's previous questions.

Yes, I believe I was using the Pulse Width trigger, and IIRC, I later confirmed that the performance suffered when I configured it back into this mode (or at least a more complex trigger mode than simple edge).

Unfortunately I've not used GLScopeClient for some time as I ran into a series of issues with getting a new version built. I do want to come back to it, but haven't been in a position to do so recently (*sigh*).

I'll close for the moment, but if I run into it again I'll re-open and try to provide more info.

closed 2021-04-25 by attie