ngscopeclient/scopehal-apps #297: Allow preferences to be reset to default values

azonenberg, 2020-12-10:
It's very plausible that a user might screw up a setting bad enough to make the software difficult or impossible to use, and not be able to figure out where they went wrong.

Also, for testing it's nice to be able to run in a default configuration and make sure things look good.

We should have the ability to reset any preference, all preferences on the current page, or all preferences to their default values.

nshcat, 2020-12-10:
Sounds good, maybe it should be placed in a right-click context menu in the preference Dialog?

azonenberg, 2020-12-10:
That would work. I don't have strong opinions on how it should be accessed, I just think it's a feature we need to have.

azonenberg, 2026-01-14:
Fixed in #945