ngscopeclient/scopehal-apps #300: December 19th hackathon meta-issue

azonenberg, 2020-12-13:
Goals and priorities for the December 19th hackathon.

### What people are working on
* @azonenberg: implement write queueing (https://github.com/azonenberg/scopehal/issues/179)

### High priority to-do's
* *Documentation*
Expand section 14 (filters) of manual. We have ~90 filters. Only 9 are fully documented. 6 more have partial documentation, and the rest are just headings or sentence of text describing the high level function of the filter.
* *Protocol Decodes*
  * Verify checksums in CAN https://github.com/azonenberg/scopehal/issues/333
  * Verify ECC in MIPI DSI https://github.com/azonenberg/scopehal/issues/328
  * Verify FCS in Ethernet https://github.com/azonenberg/scopehal/issues/55
* *DSP*
  * Implement Fibre Channel golden PLL https://github.com/azonenberg/scopehal/issues/163
  * Implement decision feedback equalizer https://github.com/azonenberg/scopehal/issues/210
* *Driver support*
  * Thorough stress testing of Rigol MSO5000 and Tek MSO5/6 driver
  * Add support for more trigger types on MSO5000
  * MSO5000 timebase/scale are wrong https://github.com/azonenberg/scopehal/issues/381
  * MSO5000 multimeter support https://github.com/azonenberg/scopehal/issues/310
  * Fix bitrotted/incomplete R&S scope driver https://github.com/azonenberg/scopehal/issues/59
* * Interchange file formats*
  * Rigol MSO5000 binary waveform import #287
  * WAV import https://github.com/azonenberg/scopehal/issues/226
  * VCD import https://github.com/azonenberg/scopehal/issues/228
* *Fix bugs/crashes*
  * Windows RGBA/BGRA swap #299
  * GLEW fails to init under Wayland #277
  * Fullscreen mode is broken on Windows #273
* *Build/packaging*
  * Get Windows CI builds to run unit tests #250
  * Make CI builds log status to IRC #247
  * Make CI builds run static analysis #240
  * Fix any remaining static analysis config problems #180
  * Make "make install" work, on both Linux and Windows #142
  * Set up initial binary release packaging using CPack #141
* *Art*
  * We need a logo/app icon. Maybe come up with a catchier name than "glscopeclient" too. #204

pd0wm, 2020-12-15:
Is it possible to contribute without compatible hardware? For example by loading a waveform from a file? I couldn't get the demo (complained about the transport being null), or siggen (only loads a hardcoded path) drivers working.

Wouldn't be interested to look into https://github.com/azonenberg/scopehal/issues/333 to learn the codebase.

azonenberg, 2020-12-15:
Yes, you can work with saved files.

Here's a test waveform of some CAN traffic with the decode already set up for you: https://www.antikernel.net/temp/glscopeclient-candemo.zip. You can open it with "glscopeclient can.scopesession" or from the File | Open menu once glscopeclient is running. There's no way to open a file from the startup dialog yet, although we plan to fix that at some point.

The demo should work also, try "glscopeclient scopename:demo:null:blah" on the command line. Due to quirks of argument parsing you need to specify an argument to the NULL transport, that was probably your problem. If you can't get the demo working, open a separate ticket or join the IRC and we'll troubleshoot there.

azonenberg, 2020-12-15:
@pd0wm Additionally, both #226 (WAV import) and #228 (VCD import) can be done without any actual hardware. I can provide test VCD files, and I'm sure you have audio files sitting around you can convert to WAV.

The CSV import code is probably a good reference point for that.

closed 2020-12-20 by azonenberg