ngscopeclient/scopehal-apps #303: glscopeclient chrome not drawn after changing applications

adamgreig, 2020-12-23:
I'm using glscopeclient 81fa6c6009a on Ubuntu 20.04 with an AMD Radeon RX550:

```
OpenGL renderer string: Radeon RX550/550 Series (POLARIS12, DRM 3.35.0, 5.4.0-58-generic, LLVM 10.0.0)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 20.0.8
OpenGL core profile shading language version string: 4.60
```

Occasionally when swapping between virtual desktops and back to glscopeclient, only the currently-updated parts of the GUI are drawn, and the rest show whatever window was previously active:

[image]

Minimising and re-showing glscopeclient or resizing its window triggers a full redraw.

azonenberg, 2021-04-24:
I think I've seen this once or twice on Linux after using OBS to do screen recordings. Unsure if same root cause or not.

azonenberg, 2022-10-03:
Wontfix in glscopeclient, should be fixed in ngscopeclient since we're redrawing every frame at 60 Hz. Reopen if we see it in ngscopeclient.

closed 2022-10-03 by azonenberg