ngscopeclient/scopehal-apps #306: Proposal migrate code from gtkmm to ImGui

bvernoux, 2021-01-05:
The idea is to migrate from gtkmm to ImGui (https://github.com/ocornut/imgui)
- ImGui is "bloat-free graphical user interface library for C++" and very fast mainly for games but it is also used for lot of amazing tools see .
- ImGui is supported by major actors like Blizzard, Google, Nvidia, Ubisoft
- ImGui license is MIT which is really better than GPL/LGPL for gtkmm

For following reasons/known issues with Gtk:
- Full-screen broken on Windows see https://github.com/azonenberg/scopehal-apps/issues/273
- gtkmm is really bloated with tons of dlls (more than 20 dependencies/dll and a real nightmare to package it and build it as static build with GPL/LGPL license conflicts ...)
- So far to display text it is not natively managed in gtkmm with opengl support and it is done with the bloated/slow cairo ... (it is natively supported with very nice/fast GUI/Widget... in ImGui)

azonenberg, 2021-01-05:
I'm not in favor at this time. Can imGui display popup windows like we use for protocol analyzers, etc? GTK for external UI chrome makes sense, it's separate from the waveform rendering and doesn't have to be super fast.

It seems like it might be a viable solution for the WaveformArea overlays, but most of that isn't performance critical anyway. For the most part, text isn't a bottleneck in performance since not much is on screen at a time. Broken fullscreen is an issue but I don't think it's worth switching UI toolkits to fix.

gtkmm has pre-existing installers and packaging, why would we build it ourselves? And if we use prebuilt binary DLL/SO packages without modifying it, LGPL vs permissive license is of no consequence.

bvernoux, 2021-01-05:
ImGui support popup window of course (including dialog ...) and tons of things much more advanced and easy to use than gtkmm (see https://github.com/ocornut/imgui/issues/1453)
I created that issue as I think for long term gtkmm is a dead end (everything is very complex for nothing and the widget/UI sutff are ugly and slow when compared to widget/UI on ImGui which is in addition heavily used in industry (mainly game but also ultra advanced tools...)

azonenberg, 2021-01-05:
We'll keep this issue open and think about it for the long term. I'm not ruling it out, but it would be a massive refactoring and I'm in no hurry.

bvernoux, 2021-01-09:
Yes there is no hurry and the advantage of ImGui is we can replace some dialog/popup/menu... from gtkmm to ImGui/ImPlot step by step without removing gtkmm (and keep gtkmm until the end if we see major improvement with ImGui/ImPlot especially with docking/multi-viewport branch)

azonenberg, 2022-09-10:
Closing this issue now that the ngscopeclient experiment is underway. We can file separate tickets for individual bugs/features in ngscopeclient as needed.

closed 2022-09-10 by azonenberg