ngscopeclient/scopehal-apps #309: Creating an AUR package for this repository

sfmth, 2021-03-19:
In Arch Linux, AUR contains lots of applications and users can submit their own. If we were to submit glscope for AUR all Arch users will be able to install this app real easy, and it would help the expansion of this app.

azonenberg, 2021-03-19:
If somebody wants to start work on packaging for distros, Arch included, that would be fine by me.

But I don't think it's stable/complete enough to be actually submitted for a main distro package repo just yet. Needs a few more months of work on stability and documentation at least.

sfmth, 2021-03-19:
I don't think if AUR requires any stability you can just provide a git repo and users compile themselves. You introduced me to this repo on twitter, I'm just getting started, and as I use Arch I could work on Arch compatibility first.
I will start working on this issue then.

azonenberg, 2021-03-19:
Yeah by all means work on packaging, since that has to be done eventually.

I just feel like the act of submitting a package to a distro implies some semblance of "usable for general audiences other than people who are active developers of the tool". As a minimum I need to fix some file format issues and make sure there's no known crashes.

602p, 2022-04-20:
FYI, someone appears to have packaged this to AUR already: https://aur.archlinux.org/packages/scopehal-apps-git . The `-git` suffix indicates it will build from the latest VCS snapshot, and that it shouldn't be taken to be stable.