ngscopeclient/scopehal-apps #325: GPU hang on

fsedano, 2021-04-28:
Getting frequent GPU hangs on Iris plus driver, with Ubuntu 20:

```
ontext: OpenGL 4.6 compatibility profile
    GL_VENDOR                   = Intel
    GL_RENDERER                 = Mesa Intel(R) Iris(R) Plus Graphics 655 (CFL GT3)
    GL_VERSION                  = 4.6 (Core Profile) Mesa 20.2.6
    GL_SHADING_LANGUAGE_VERSION = 4.60
    Initial GL error code       = 0
    GL_ARB_gpu_shader_int64     = supported

```
Demo at:
[gpu hang](https://youtu.be/5vFP06t51SU)

Capture of the hang:

https://cisco.box.com/s/frr2mo8sww7yq79csewltegc4y2rfz5k

azonenberg, 2021-04-30:
Suspect it's related to #327 but keeping this ticket open until we've confirmed that's the case.

azonenberg, 2021-05-03:
Fixed by https://github.com/azonenberg/scopehal/pull/450

fsedano, 2021-05-07:
Hi - This happened again even after the fix. Something else is going on. Same signature...

mandl, 2022-01-28:
Hello Team,

same on my Ubuntu 20.

Context: OpenGL 4.6 compatibility profile
    GL_VENDOR                   = Intel
    GL_RENDERER               = Mesa Intel(R) UHD Graphics (CML GT2)
    GL_VERSION                  = 4.6 (Core Profile) Mesa 21.0.3
    GL_SHADING_LANGUAGE_VERSION = 4.60
    Initial GL error code       = 0
    GL_ARB_gpu_shader_int64     = supported
Requested rendering backend: OpenGL
Using rendering backend: OpenGL

[ 2169.694889] i915 0000:00:02.0: [drm] Resetting rcs0 for preemption time out
[ 2169.694914] i915 0000:00:02.0: [drm] glscopeclient[9966] context reset due to GPU hang
[ 2169.700445] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:1:8ed8fff3, in glscopeclient [9966]

mandl, 2022-01-28:
Workaround - use the old driver

`MESA_LOADER_DRIVER_OVERRIDE=i965 glscopeclient `

https://linuxreviews.org/Intel_Iris

azonenberg, 2023-03-09:
Is anyone still seeing this issue with recent builds and/or ngscopeclient?

smunaut, 2023-03-13:
FWIW:

```
[9485860.755044] i915 0000:00:02.0: GPU HANG: ecode 7:1:0xfefffffe, in ngscopeclient [13528], hang on rcs0
[9485860.755087] i915 0000:00:02.0: Resetting chip for hang on rcs0
```

But it's on a Haswell with incomplete vulkan support.

azonenberg, 2023-12-06:
Has anyone seen Intel driver issues with a relatively recent build? I know we fixed some shader bugs over the summer. Would like to close this if it's not still happening.

smunaut, 2023-12-06:
It's still there, just reproduced it instantly in a build of master right now.
Running on intel 12th gen iGPU with Mesa 23.3.0

commit 0f265a9d (2024-01-28): Waveform rendering shader: Group write to termination flag

Instead of writing to the termination flag from different
point in the code, which might lead to some writes being
issued while there are already pending writes from other
threads coming from other code path.

This seems to be an issue on intel at least.

So instead each thread keeps a local flag and then in
the common execution path, that flag might trigger a write
to the shared variable and immediately after a barrier will
be issued before any work item uses the flag or try to issue
another write to it.

Fixes #325
closed 2024-01-28 by azonenberg