ngscopeclient/scopehal-apps #361: When cycling back through the history, trigger point and channel are not updated

mubes, 2021-05-28:
Steps to reproduce; Record waveform A using trigger x, Record waveform B using trigger y. Use history to go back to waveform A, trigger y is still displayed.

azonenberg, 2021-05-28:
All scope settings (vertical scale, trigger configuration, etc) are displayed at their current values. Historical waveforms only store sample data, not a full snapshot of scope settings.

602p, 2022-05-19:
Having a snapshot of scope settings would be useful for me. We should maybe think generally about how to save scope setup information like this in historical waveforms. Function generator state will be another thing like this, since it would be nice to be able to e.g. characterize a filter at a range of inputs and save the whole thing as a .scopesession that includes the input func gen settings for each capture too.

This issue, #283, #343, #360 in my mind form kind of a feature bundle that would allow a scopesession to be "notebook-like" in containing labeled captures with their at-capture scope configs. Maybe open tracker ticket for that if that jives with your vision?

azonenberg, 2022-05-19:
Yes, I think that would be useful.