ngscopeclient/scopehal-apps #376: std::bad_alloc exception when running the demo scope for some time

electroniceel, 2021-06-06:
How to reproduce:
- glscopeclient --noopencl
- File > Connect, Driver demo, Transport null
- Delete all waveforms except the Ramp
- Right-Click into the Ramp waveform, Alphabetical, FFT
- Leave running for a minute or two

-> Program aborts with
```
unhandled exception (type std::exception) in signal handler:
what: std::bad_alloc
```
gdb backtrace:
```
#0  0x00007ffff6d159e5 in raise () at /lib64/libc.so.6
#1  0x00007ffff6cfe8a4 in abort () at /lib64/libc.so.6
#2  0x00007ffff70a4926 in __gnu_cxx::__verbose_terminate_handler() [clone .cold] () at /lib64/libstdc++.so.6
#3  0x00007ffff70b008c in __cxxabiv1::__terminate(void (*)()) () at /lib64/libstdc++.so.6
#4  0x00007ffff70af119 in  () at /lib64/libstdc++.so.6
#5  0x00007ffff70afa61 in __gxx_personality_v0 () at /lib64/libstdc++.so.6
#6  0x00007ffff6eb3bff in _Unwind_RaiseException_Phase2 () at /lib64/libgcc_s.so.1
#7  0x00007ffff6eb4291 in _Unwind_RaiseException () at /lib64/libgcc_s.so.1
#8  0x00007ffff70b039c in __cxa_throw () at /lib64/libstdc++.so.6
#9  0x0000555555611f8e in AlignedAllocator<EmptyConstructorWrapper<long>, 64ul>::allocate(unsigned long) const (this=0x7ffb77fb7c78, n=<optimized out>)
    at /usr/src/debug/scopehal-apps-0.0.99-20210602.ceb14ff.fc33.x86_64/lib/scopehal/AlignedAllocator.h:127
#10 std::allocator_traits<AlignedAllocator<EmptyConstructorWrapper<long>, 64ul> >::allocate(AlignedAllocator<EmptyConstructorWrapper<long>, 64ul>&, unsigned long) (__n=<optimized out>, __a=...)
    at /usr/include/c++/10/bits/alloc_traits.h:314
#11 std::_Vector_base<EmptyConstructorWrapper<long>, AlignedAllocator<EmptyConstructorWrapper<long>, 64ul> >::_M_allocate(unsigned long) (__n=<optimized out>, this=<optimized out>)
    at /usr/include/c++/10/bits/stl_vector.h:346
#12 std::vector<EmptyConstructorWrapper<long>, AlignedAllocator<EmptyConstructorWrapper<long>, 64ul> >::_M_default_append(unsigned long) (this=0x7ffb77fb7c78, __n=100000)
    at /usr/include/c++/10/bits/vector.tcc:635
#13 0x00007ffff73fb91f in std::vector<EmptyConstructorWrapper<long>, AlignedAllocator<EmptyConstructorWrapper<long>, 64ul> >::resize(unsigned long) (__new_size=100000, this=0x7ffb77fb7c78)
    at /usr/include/c++/10/bits/stl_vector.h:940
#14 Waveform<EmptyConstructorWrapper<float> >::Resize(unsigned long) (size=100000, this=0x7ffb77fb7c30)
    at /usr/src/debug/scopehal-apps-0.0.99-20210602.ceb14ff.fc33.x86_64/lib/scopehal/Waveform.h:181
#15 TestWaveformSource::GenerateNoisySinewave(float, float, float, long, unsigned long, float)
    (this=0x555556ab5fa0, amplitude=0.899999976, startphase=0, period=1000000, sampleperiod=20000, depth=100000, noise_amplitude=0.00999999978)
    at /usr/src/debug/scopehal-apps-0.0.99-20210602.ceb14ff.fc33.x86_64/lib/scopehal/TestWaveformSource.cpp:119
#16 0x00007ffff7364f9e in DemoOscilloscope::AcquireData() [clone ._omp_fn.0] () at /usr/src/debug/scopehal-apps-0.0.99-20210602.ceb14ff.fc33.x86_64/lib/scopehal/DemoOscilloscope.cpp:486
#17 0x00007ffff7222d36 in GOMP_parallel () at /lib64/libgomp.so.1
#18 0x00007ffff73695f1 in DemoOscilloscope::AcquireData() () at /usr/src/debug/scopehal-apps-0.0.99-20210602.ceb14ff.fc33.x86_64/lib/scopehal/DemoOscilloscope.cpp:480
#19 0x000055555567a725 in ScopeThread(Oscilloscope*) (scope=0x555556a8f300) at /usr/src/debug/scopehal-apps-0.0.99-20210602.ceb14ff.fc33.x86_64/src/glscopeclient/main.cpp:292
#20 0x00007ffff70dc444 in execute_native_thread_routine () at /lib64/libstdc++.so.6
#21 0x00007ffff71f73f9 in start_thread () at /lib64/libpthread.so.0
#22 0x00007ffff6dd94c3 in clone () at /lib64/libc.so.6
```

Platform: Fedora 33 on x86_64 with my RPM-Packages.
Version: ceb14ff from 2021-06-02

azonenberg, 2021-06-07:
Having trouble reproducing on my Debian system. Wonder if it's something specific to your build... Might also be something racey (yay debugging threaded code)

electroniceel, 2021-06-17:
I just rebuilt the current version (b327de0 from today) and can't reproduce it now too. It could have been fixed without recognizing it, but my guess is more that the bug showing up depends on exact memory positions of some functions in the binary or in the generated data and is still there. 

This would also explain why you couldn't reproduce it on Debian, as the library versions, compiler version and compile and link options differ a bit from my Fedora system.

azonenberg, 2022-10-23:
Closing as not reproducible for now, will reopen if anyone sees it in the future.

closed 2022-10-23 by azonenberg