ngscopeclient/scopehal-apps #384: ERROR: OpenCL error: clBuildProgram (-11)

hei5enbrg, 2021-08-10:
Trying to get glscopeclient running on Manjaro Linux. After installation, when trying to launch I'm getting 

`ERROR: OpenCL error: clBuildProgram (-11)
ERROR: Failed to build OpenCL program for FFT
[1]    20465 segmentation fault (core dumped)  glscopeclient`

azonenberg, 2021-08-11:
It looks like you have two separate issues going on...

1) The FFT filter is failing to initialize the OpenCL back end for GPU acceleration. This is probably a GPU driver issue, but it's hard to be sure without more information.

2) This failure is causing glscopeclient to crash rather than fall back to non-accelerated FFTs. This is a bug in libscopehal which I'll investigate tonight.

Can you post the output of "glscopeclient --debug" here?

azonenberg, 2021-08-11:
Found and fixed a bug in the error handling code that might be causing the crash (problem 2). Please update to latest HEAD (be sure to pull submodules as well) and let me know if the crash is gone.

This won't fix the OpenCL startup so you still won't get accelerated FFTs, but it should at least not segfault anymore.

hei5enbrg, 2021-08-14:
Fixed the issue after installing `opencl-amd` from AUR, without that package glscopeclient is crashing with the new HEAD as well. Attaching the logs 
```
$glscopeclient --debug

Detecting CPU features...
    * AVX2
    * FMA
    
Detecting OpenCL devices...
    Platform 0
        CL_PLATFORM_NAME    = Clover
        CL_PLATFORM_PROFILE = FULL_PROFILE
        CL_PLATFORM_VENDOR  = Mesa
        CL_PLATFORM_VERSION = OpenCL 1.1 Mesa 21.1.6
        Device 0
            CL_DRIVER_VERSION                   = 21.1.6
            CL_DEVICE_NAME                      = AMD Radeon RX 5700 XT (NAVI10, DRM 3.40.0, 5.10.56-1-MANJARO, LLVM 12.0.1)
            CL_DEVICE_OPENCL_C_VERSION          = OpenCL C 1.1 
            CL_DEVICE_PROFILE                   = FULL_PROFILE
            CL_DEVICE_VENDOR                    = AMD
            CL_DEVICE_VERSION                   = OpenCL 1.1 Mesa 21.1.6
            CL_DEVICE_GLOBAL_MEM_CACHE_SIZE     = 0.000 MB
            CL_DEVICE_GLOBAL_MEM_CACHELINE_SIZE = 140733193388032
            CL_DEVICE_GLOBAL_MEM_SIZE           = 8.00 GB
            CL_DEVICE_LOCAL_MEM_SIZE            = 32.00 kB
            CL_DEVICE_MAX_CLOCK_FREQUENCY       = 2100 MHz
            CL_DEVICE_MAX_COMPUTE_UNITS         = 40
            CL_DEVICE_MAX_CONSTANT_ARGS         = 16
            CL_DEVICE_MAX_CONSTANT_BUFFER_SIZE  = 65536.00 kB
            CL_DEVICE_MAX_MEM_ALLOC_SIZE        = 6.40 GB
            CL_DEVICE_MAX_PARAMETER_SIZE        = 1024
            CL_DEVICE_MAX_WORK_GROUP_SIZE       = 256
            CL_DEVICE_MAX_WORK_ITEM_SIZES       = 256, 256, 256
            CL_DEVICE_EXTENSIONS:
                cl_khr_byte_addressable_store
                cl_khr_global_int32_base_atomics
                cl_khr_global_int32_extended_atomics
                cl_khr_local_int32_base_atomics
                cl_khr_local_int32_extended_atomics
                cl_khr_int64_base_atomics
                cl_khr_int64_extended_atomics
                cl_khr_fp64
clFFT version: 2.12.2

ERROR: OpenCL error: clBuildProgram (-11)
ERROR: Failed to build OpenCL program for FFT
Window program build log:
fatal error: cannot open file '/usr/share/clc/gfx1010-amdgcn-mesa-mesa3d.bc': No such file or directory

[1]    187703 segmentation fault (core dumped)  glscopeclient --debug
```
After installing `opencl-amd` 
```
$glscopeclient --debug

Detecting CPU features...
    * AVX2
    * FMA
    
Detecting OpenCL devices...
    Platform 0
        CL_PLATFORM_NAME    = AMD Accelerated Parallel Processing
        CL_PLATFORM_PROFILE = FULL_PROFILE
        CL_PLATFORM_VENDOR  = Advanced Micro Devices, Inc.
        CL_PLATFORM_VERSION = OpenCL 2.0 AMD-APP (3314.0)
        Device 0
            CL_DRIVER_VERSION                   = 3314.0 (HSA1.1,LC)
            CL_DEVICE_NAME                      = gfx1010:xnack-
            CL_DEVICE_OPENCL_C_VERSION          = OpenCL C 2.0 
            CL_DEVICE_PROFILE                   = FULL_PROFILE
[... 44 lines cut ...]
            CL_DEVICE_VENDOR                    = AMD
            CL_DEVICE_VERSION                   = OpenCL 1.1 Mesa 21.1.6
            CL_DEVICE_GLOBAL_MEM_CACHE_SIZE     = 0.000 MB
            CL_DEVICE_GLOBAL_MEM_CACHELINE_SIZE = 0
            CL_DEVICE_GLOBAL_MEM_SIZE           = 8.00 GB
            CL_DEVICE_LOCAL_MEM_SIZE            = 32.00 kB
            CL_DEVICE_MAX_CLOCK_FREQUENCY       = 2100 MHz
            CL_DEVICE_MAX_COMPUTE_UNITS         = 40
            CL_DEVICE_MAX_CONSTANT_ARGS         = 16
            CL_DEVICE_MAX_CONSTANT_BUFFER_SIZE  = 65536.00 kB
            CL_DEVICE_MAX_MEM_ALLOC_SIZE        = 6.40 GB
            CL_DEVICE_MAX_PARAMETER_SIZE        = 1024
            CL_DEVICE_MAX_WORK_GROUP_SIZE       = 256
            CL_DEVICE_MAX_WORK_ITEM_SIZES       = 256, 256, 256
            CL_DEVICE_EXTENSIONS:
                cl_khr_byte_addressable_store
                cl_khr_global_int32_base_atomics
                cl_khr_global_int32_extended_atomics
                cl_khr_local_int32_base_atomics
                cl_khr_local_int32_extended_atomics
                cl_khr_int64_base_atomics
                cl_khr_int64_extended_atomics
                cl_khr_fp64
clFFT version: 2.12.2
```

azonenberg, 2021-08-14:
Looks like I had a typo in a copy-pasted error message where the de-embedding filter called itself FFT in one of the error lines. I had the same null dereference there, but was looking for it in the wrong place...

Can you uninstall `opencl-amd` temporarily, confirm the crash again, then update to latest HEAD and see if it's fixed?

hei5enbrg, 2021-08-15:
I am able to reproduce the crash without `opencl-amd`. With the latest HEAD, without `opencl-amd` the program is working fine, though I observed a crash when trying to load FFT module from `right click->RF->FFT`

With the latest HEAD, without `opencl-amd`
```
$glscopeclient --debug

Detecting CPU features...
    * AVX2
    * FMA
    
Detecting OpenCL devices...
    Platform 0
        CL_PLATFORM_NAME    = Clover
        CL_PLATFORM_PROFILE = FULL_PROFILE
        CL_PLATFORM_VENDOR  = Mesa
        CL_PLATFORM_VERSION = OpenCL 1.1 Mesa 21.1.6
        Device 0
            CL_DRIVER_VERSION                   = 21.1.6
            CL_DEVICE_NAME                      = AMD Radeon RX 5700 XT (NAVI10, DRM 3.40.0, 5.10.56-1-MANJARO, LLVM 12.0.1)
            CL_DEVICE_OPENCL_C_VERSION          = OpenCL C 1.1 
            CL_DEVICE_PROFILE                   = FULL_PROFILE
            CL_DEVICE_VENDOR                    = AMD
            CL_DEVICE_VERSION                   = OpenCL 1.1 Mesa 21.1.6
            CL_DEVICE_GLOBAL_MEM_CACHE_SIZE     = 0.000 MB
            CL_DEVICE_GLOBAL_MEM_CACHELINE_SIZE = 140733193388032
            CL_DEVICE_GLOBAL_MEM_SIZE           = 8.00 GB
            CL_DEVICE_LOCAL_MEM_SIZE            = 32.00 kB
            CL_DEVICE_MAX_CLOCK_FREQUENCY       = 2100 MHz
            CL_DEVICE_MAX_COMPUTE_UNITS         = 40
            CL_DEVICE_MAX_CONSTANT_ARGS         = 16
            CL_DEVICE_MAX_CONSTANT_BUFFER_SIZE  = 65536.00 kB
            CL_DEVICE_MAX_MEM_ALLOC_SIZE        = 6.40 GB
            CL_DEVICE_MAX_PARAMETER_SIZE        = 1024
            CL_DEVICE_MAX_WORK_GROUP_SIZE       = 256
            CL_DEVICE_MAX_WORK_ITEM_SIZES       = 256, 256, 256
            CL_DEVICE_EXTENSIONS:
                cl_khr_byte_addressable_store
                cl_khr_global_int32_base_atomics
                cl_khr_global_int32_extended_atomics
                cl_khr_local_int32_base_atomics
                cl_khr_local_int32_extended_atomics
                cl_khr_int64_base_atomics
                cl_khr_int64_extended_atomics
                cl_khr_fp64
clFFT version: 2.12.2

ERROR: OpenCL error: clBuildProgram (-11)
ERROR: Failed to build OpenCL program for de-embed
Window program build log:
fatal error: cannot open file '/usr/share/clc/gfx1010-amdgcn-mesa-mesa3d.bc': No such file or directory

ERROR: OpenCL error: clBuildProgram (-11)
ERROR: Failed to build OpenCL program for de-embed
Window program build log:
fatal error: cannot open file '/usr/share/clc/gfx1010-amdgcn-mesa-mesa3d.bc': No such file or directory

ERROR: OpenCL error: clBuildProgram (-11)
ERROR: Failed to build OpenCL program for de-embed
Window program build log:
fatal error: cannot open file '/usr/share/clc/gfx1010-amdgcn-mesa-mesa3d.bc': No such file or directory

ERROR: OpenCL error: clBuildProgram (-11)
ERROR: Failed to build OpenCL program for FFT
Window program build log:
fatal error: cannot open file '/usr/share/clc/gfx1010-amdgcn-mesa-mesa3d.bc': No such file or directory

Normalize program object not present
ERROR: OpenCL error: clBuildProgram (-11)
ERROR: Failed to build OpenCL program from kernels/FIRFilter.cl
Build log:
[... 277 lines cut ...]

ERROR: OpenCL error: clBuildProgram (-11)
ERROR: Failed to build OpenCL program for FFT
Window program build log:
fatal error: cannot open file '/usr/share/clc/gfx1010-amdgcn-mesa-mesa3d.bc': No such file or directory

Normalize program object not present
ERROR: OpenCL error: clBuildProgram (-11)
ERROR: Failed to build OpenCL program for FFT
Window program build log:
fatal error: cannot open file '/usr/share/clc/gfx1010-amdgcn-mesa-mesa3d.bc': No such file or directory

Normalize program object not present

                        BUILD LOG
************************************************
fatal error: cannot open file '/usr/share/clc/gfx1010-amdgcn-mesa-mesa3d.bc': No such file or directory

************************************************
FFTGeneratedTransposeGCNAction::compileKernels failed
ERROR: clfftBakePlan failed (-11)
[1]    226021 abort (core dumped)  glscopeclient --debug
```

Thanks for the help. I came here after watching your hardwear.io talk :)

azonenberg, 2021-08-15:
Interesting, so it looks like my new error handling is working and it's crashing further down. This log isn't enough for me to diagnose.

Can you rebuild with debug symbols (cmake -DCMAKE_BUILD_TYPE=DEBUG) and get me a backtrace of the crash?

hei5enbrg, 2021-08-15:
`cmake -DCMAKE_BUILD_TYPE=DEBUG` did not add any additional info to the logs. 

Logs from `lldb glscopeclient --debug'
```
                        BUILD LOG
************************************************
fatal error: cannot open file '/usr/share/clc/gfx1010-amdgcn-mesa-mesa3d.bc': No such file or directory

************************************************
FFTGeneratedTransposeGCNAction::compileKernels failed
E
R
R
O
R
:
 
c
l
f
f
t
B
a
k
e
P
l
a
n
 
f
a
i
l
e
d
 
(
-
1
1
)

error: glscopeclient {0x006d68b0}: DIE has DW_AT_ranges(0x1fa6a) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: glscopeclient {0x006d68b0}: DIE has DW_AT_ranges(0x1fa6a) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x0002d69b}: DIE has DW_AT_ranges(0x33c9) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037cc3}: DIE has DW_AT_ranges(0x38d7) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037e02}: DIE has DW_AT_ranges(0x38fb) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037cc3}: DIE has DW_AT_ranges(0x38d7) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037cf6}: DIE has DW_AT_ranges(0x38e5) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037d1d}: DIE has DW_AT_ranges(0x38e5) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037e02}: DIE has DW_AT_ranges(0x38fb) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037e31}: DIE has DW_AT_ranges(0x390b) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037e64}: DIE has DW_AT_ranges(0x390b) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037f1a}: DIE has DW_AT_ranges(0x3924) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037f49}: DIE has DW_AT_ranges(0x3924) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037cc3}: DIE has DW_AT_ranges(0x38d7) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037e02}: DIE has DW_AT_ranges(0x38fb) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037cc3}: DIE has DW_AT_ranges(0x38d7) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037e02}: DIE has DW_AT_ranges(0x38fb) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037cc3}: DIE has DW_AT_ranges(0x38d7) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037e02}: DIE has DW_AT_ranges(0x38fb) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037cc3}: DIE has DW_AT_ranges(0x38d7) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037e02}: DIE has DW_AT_ranges(0x38fb) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037cc3}: DIE has DW_AT_ranges(0x38d7) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: libgomp.so.1 {0x00037e02}: DIE has DW_AT_ranges(0x38fb) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
[... cut ...]
error: libgomp.so.1 {0x00037e02}: DIE has DW_AT_ranges(0x38fb) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
Process 239610 stopped
* thread #1, name = 'glscopeclient', stop reason = signal SIGABRT
    frame #0: 0x00007ffff5969d22 libc.so.6`raise + 322
libc.so.6`raise:
->  0x7ffff5969d22 <+322>: movq   0x108(%rsp), %rax
    0x7ffff5969d2a <+330>: subq   %fs:0x28, %rax
    0x7ffff5969d33 <+339>: jne    0x7ffff5969d54            ; <+372>
    0x7ffff5969d35 <+341>: movl   %r8d, %eax
(lldb) 
```

azonenberg, 2021-08-15:
Enabling debug during compilation doesn't change the printed output, it just means you get less-optimized binaries with source level debug symbols (-O0 -g).

Can I get a stack trace?

hei5enbrg, 2021-08-15:
```
error: libgomp.so.1 {0x00037649}: DIE has DW_AT_ranges(0x3994) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
Process 242617 stopped
* thread #1, name = 'glscopeclient', stop reason = signal SIGABRT
    frame #0: 0x00007ffff5969d22 libc.so.6`raise + 322
libc.so.6`raise:
->  0x7ffff5969d22 <+322>: movq   0x108(%rsp), %rax
    0x7ffff5969d2a <+330>: subq   %fs:0x28, %rax
    0x7ffff5969d33 <+339>: jne    0x7ffff5969d54            ; <+372>
    0x7ffff5969d35 <+341>: movl   %r8d, %eax
(lldb) bt
error: glscopeclient {0x00316102}: DIE has DW_AT_ranges(0x8f71) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: glscopeclient {0x004c7206}: DIE has DW_AT_ranges(0x19826) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: glscopeclient {0x00829bdf}: DIE has DW_AT_ranges(0x279ee) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
error: glscopeclient {0x00829c30}: DIE has DW_AT_ranges(0x27a14) attribute, but range extraction failed (missing or invalid range list table), please file a bug and attach the file at the start of this error message
* thread #1, name = 'glscopeclient', stop reason = signal SIGABRT
  * frame #0: 0x00007ffff5969d22 libc.so.6`raise + 322
    frame #1: 0x00007ffff5953862 libc.so.6`abort + 278
    frame #2: 0x00007ffff7e4fab9 libscopeprotocols.so`FFTFilter::ReallocateBuffers(this=0x0000555555c57400, npoints_raw=100000, npoints=131072, nouts=65537) at FFTFilter.cpp:291:11
    frame #3: 0x00007ffff7e4fc03 libscopeprotocols.so`FFTFilter::Refresh(this=0x0000555555c57400) at FFTFilter.cpp:321:20
    frame #4: 0x00005555556f5c73 glscopeclient`WaveformArea::OnDecodeSetupComplete(this=0x0000555555ea0000) at WaveformArea_events.cpp:1169:26
    frame #5: 0x00005555556f5702 glscopeclient`WaveformArea::OnProtocolDecode(this=0x0000555555ea0000, name="FFT", forceStats=false) at WaveformArea_events.cpp:1089:24
    frame #6: 0x00005555556f12b4 glscopeclient`sigc::bound_mem_functor2<void, WaveformArea, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool>::operator(this=0x00005555560a8770, _A_a1=0x00005555560a8790, _A_a2=0x00005555560a87b0)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool const&) const at mem_fun.h:2143:48
    frame #7: 0x00005555556f0e7b glscopeclient`sigc::adaptor_functor<sigc::bound_mem_functor2<void, WaveformArea, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool> >::deduce_result_type<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, bool&, void, void, void, void, void>::type sigc::adaptor_functor<sigc::bound_mem_functor2<void, WaveformArea, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool> >::operator(this=0x00005555560a8768, _A_arg1=0x00005555560a8790, _A_arg2=0x00005555560a87b0)<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, bool&>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, bool&) const at adaptor_trait.h:108:22
    frame #8: 0x00005555556f095a glscopeclient`sigc::bind_functor<-1, sigc::bound_mem_functor2<void, WaveformArea, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil>::operator(this=0x00005555560a8760)() at bind.h:1340:186
    frame #9: 0x00005555556f0026 glscopeclient`sigc::internal::slot_call0<sigc::bind_functor<-1, sigc::bound_mem_functor2<void, WaveformArea, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil>, void>::call_it(rep=0x00005555560a8730) at slot.h:136:35
    frame #10: 0x00007ffff6a48e49 libglibmm-2.4.so.1`Glib::SignalProxyNormal::slot0_void_callback(_GObject*, void*) + 41
    frame #11: 0x00007ffff66ced8f libgobject-2.0.so.0`g_closure_invoke + 367
    frame #12: 0x00007ffff66ea77b libgobject-2.0.so.0`___lldb_unnamed_symbol273$$libgobject-2.0.so.0 + 3195
    frame #13: 0x00007ffff66ebc49 libgobject-2.0.so.0`g_signal_emit_valist + 3705
    frame #14: 0x00007ffff66ec1a0 libgobject-2.0.so.0`g_signal_emit + 144
    frame #15: 0x00007ffff70d9895 libgtk-3.so.0`gtk_widget_activate + 85
    frame #16: 0x00007ffff6f9fa4d libgtk-3.so.0`gtk_menu_shell_activate_item + 333
    frame #17: 0x00007ffff6f9fd24 libgtk-3.so.0`___lldb_unnamed_symbol4423$$libgtk-3.so.0 + 612
    frame #18: 0x00007ffff6e311e5 libgtk-3.so.0`___lldb_unnamed_symbol66$$libgtk-3.so.0 + 149
    frame #19: 0x00007ffff66ec030 libgobject-2.0.so.0`g_signal_emit_valist + 4704
    frame #20: 0x00007ffff66ec1a0 libgobject-2.0.so.0`g_signal_emit + 144
    frame #21: 0x00007ffff70ecce5 libgtk-3.so.0`___lldb_unnamed_symbol7201$$libgtk-3.so.0 + 437
    frame #22: 0x00007ffff6f8a9a5 libgtk-3.so.0`___lldb_unnamed_symbol4191$$libgtk-3.so.0 + 357
[... cut ...]
    frame #43: 0x00007ffff7a523be libgtkmm-3.0.so.1`Gtk::Main::iteration(bool) + 14
    frame #44: 0x00005555556c9e9b glscopeclient`ScopeApp::run(this=0x00005555557e1b90, filesToLoad=size=0, reconnect=false, nodata=false, retrigger=false, nodigital=false, nospectrum=false) at ScopeApp.cpp:143:23
    frame #45: 0x000055555572300e glscopeclient`main(argc=1, argv=0x00007fffffffd768) at main.cpp:215:12
    frame #46: 0x00007ffff5954b25 libc.so.6`__libc_start_main + 213
    frame #47: 0x00005555555fe36e glscopeclient`_start + 46
(lldb) 
```

azonenberg, 2021-08-15:
Looks like with this particular driver breakage, I get most of the way through initializing the accelerated FFT then something fails to run. This confused the filter, which assumed that if this much OpenCL was alive, things would mostly work.

Added some fallback code that should gracefully degrade to CPU based FFTs if clFFT fails to bake the plan. Let me know if it works.

azonenberg, 2022-03-27:
Closing due to lack of activity for an extended time period.

closed 2022-03-27 by azonenberg