ngscopeclient/scopehal-apps #385: ERROR: Internal compile error, error code: E_SC_NOTSUPPORTED Shader not supported by HW

ncicek, 2021-08-14:
Repro steps:
Install AMD PRO gpu drivers on ubuntu 20 (needed for OpenCL)
Launch glscopeclient and connect to demo driver.
Program crashes and following error appears:

$ glscopeclient 
ERROR: Link of shader progam failed:
Compute shader(s) failed to link.
Compute link error: HW_UNSUPPORTED.
ERROR: Internal compile error, error code: E_SC_NOTSUPPORTED
Shader not supported by HW
INTERNAL ERROR: failed to link histogram waveform shader program, aborting
    This indicates a bug in the program, please file a report via Github
Aborted (core dumped)

Doesn't seem to matter which AMD GPU I try: old radeon 7900 series, or newer rx 6800.

Googling for keyword "E_SC_NOTSUPPORTED" shows references to SSBO (Shader Storage Buffer Object). Is this something that is used in the shader code?
How can I help debug further?

ncicek, 2021-08-14:
Here is full output when run with --debug flag:

$ glscopeclient --debug
OMP_WAIT_POLICY not set to PASSIVE. Re-exec'ing with correct environment
Detecting CPU features...
    * AVX2
    * FMA
    
Detecting OpenCL devices...
    Platform 0
        CL_PLATFORM_NAME    = AMD Accelerated Parallel Processing
        CL_PLATFORM_PROFILE = FULL_PROFILE
        CL_PLATFORM_VENDOR  = Advanced Micro Devices, Inc.
        CL_PLATFORM_VERSION = OpenCL 2.1 AMD-APP (3302.5)
        No GPUs found
clFFT version: 2.12.2

Connecting to SCPI oscilloscope at 192.168.1.39:5025
Context: OpenGL 4.2 compatibility profile
    GL_VENDOR                   = (null)
    GL_RENDERER                 = (null)
    GL_VERSION                  = (null)
    GL_SHADING_LANGUAGE_VERSION = (null)
    Initial GL error code       = 0
    GL_ARB_gpu_shader_int64     = not supported
ERROR: Link of shader progam failed:
Compute shader(s) failed to link.
Compute link error: HW_UNSUPPORTED.
ERROR: Internal compile error, error code: E_SC_NOTSUPPORTED
Shader not supported by HW
INTERNAL ERROR: failed to link histogram waveform shader program, aborting
    This indicates a bug in the program, please file a report via Github
Aborted (core dumped)

ncicek, 2021-08-14:
Here is output of clinfo on my system:

$ clinfo
Number of platforms                               1
  Platform Name                                   AMD Accelerated Parallel Processing
  Platform Vendor                                 Advanced Micro Devices, Inc.
  Platform Version                                OpenCL 2.1 AMD-APP (3302.5)
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd cl_amd_event_callback cl_amd_offline_devices 
  Platform Host timer resolution                  1ns
  Platform Extensions function suffix             AMD

  Platform Name                                   AMD Accelerated Parallel Processing
Number of devices                                 0

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  No platform
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   No platform
  clCreateContext(NULL, ...) [default]            No platform
  clCreateContext(NULL, ...) [other]              No platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  No devices found in platform

azonenberg, 2021-08-14:
This is not an OpenCL issue, this is OpenGL compute shader related. glscopeclient's renderer uses shader storage buffer objects for displaying waveforms.

Can you post glxinfo output?

ncicek, 2021-08-14:
Thank you for the quick response.
I realized that this issue doesn't happen with the default mainline driver (ie if I install amd-gpu-pro driver with --headless option, it just gives me the opencl and no driver).
So I took two glxinfo outputs attached below. One with the mainline driver (which works, no error msg) and one with amd-gpu-pro driver (gives error).
[amd-gpu-pro-driver-glxinfo.txt](https://github.com/azonenberg/scopehal-apps/files/6987237/amd-gpu-pro-driver-glxinfo.txt)
[mainline-driver-glxinfo.txt](https://github.com/azonenberg/scopehal-apps/files/6987238/mainline-driver-glxinfo.txt)

ncicek, 2021-08-15:
I think I see the cause.
On the amd-gpu-pro driver, the GL_ARB_shader_storage_buffer_object extension support is on "ES profile" however the mainline driver has this extension under "core" profile.

Is there any way to make it use the ES profile?

azonenberg, 2021-08-15:
The amd-gpu-pro driver reports GL_ARB_shader_storage_buffer_object under core profile, and also reports OpenGL 4.6 (this extension is required in core since... 4.3 I think).

Are you using the most recent version of amd-gpu-pro for your card? Does it provide any features mainline does not that you need? I'm mostly an nvidia shop so not that familiar with AMD's drivers.

ncicek, 2021-08-15:
Yes, this was tested on latest 20.30 version of amdgpu-pro driver.
I don't really need to use amdgpu-pro driver over mainline.
However, the same error comes up on Windows when I test the latest windows build (downloaded from github). On Windows there is no mainline driver alternative so that makes this more important.

Any ideas? Issue seems to be common to amd's closed-source opengl driver on both windows and linux side.

C:\Users\xxx\Desktop\glscopeclient>glscopeclient.exe --debug
←[33;1mWarning:←[0m glscopeclient works best with the OMP_WAIT_POLICY environment variable set to PASSIVE
Detecting CPU features...
    * AVX2
    * FMA

    ←[33;1mWarning:←[0m AVX2/AVX512 detected but disabled on MinGW64/GCC (see https://github.com/azonenberg/scopehal-apps/issues/29
    5)
OpenCL support: not present at compile time. GPU acceleration disabled.

Context: OpenGL 4.2 compatibility profile
    GL_VENDOR                   = ATI Technologies Inc.
    GL_RENDERER                 = AMD Radeon RX 6800
    GL_VERSION                  = 4.2.14742 Core Profile Context 21.8.1 27.20.22023.1004
    GL_SHADING_LANGUAGE_VERSION = 4.60
    Initial GL error code       = 0
    GL_ARB_gpu_shader_int64     = not supported
←[31;1mERROR:←[0m Link of shader progam failed:
Compute shader(s) failed to link.
←[31;1mCompute link error:←[0m HW_UNSUPPORTED.
←[31;1mERROR:←[0m Internal compile error, error code: E_SC_NOTSUPPORTED
Shader not supported by HW
←[31;1m←[31;1mINTERNAL ERROR:←[0m←[0m failed to link histogram waveform shader program, aborting
    This indicates a bug in the program, please file a report via Github

azonenberg, 2021-08-15:
Can you post the list of OpenGL extensions on the Windows system? I know there's third party utilities for this, not sure if AMD's software can do it natively.

Also, you can run `glscopeclient --debug --logfile foo.txt` to get output logged to a file in plain unformatted text without any of the color coding control characters.

ncicek, 2021-08-15:
Output of foo.txt log:
`Warning: glscopeclient works best with the OMP_WAIT_POLICY environment variable set to PASSIVE
Detecting CPU features...
    * AVX2
    * FMA
    
    Warning: AVX2/AVX512 detected but disabled on MinGW64/GCC (see https://github.com/azonenberg/scopehal-apps/issues/29
    5)
OpenCL support: not present at compile time. GPU acceleration disabled.

Context: OpenGL 4.2 compatibility profile
    GL_VENDOR                   = ATI Technologies Inc.
    GL_RENDERER                 = AMD Radeon RX 6800
    GL_VERSION                  = 4.2.14742 Core Profile Context 21.8.1 27.20.22023.1004
    GL_SHADING_LANGUAGE_VERSION = 4.60
    Initial GL error code       = 0
    GL_ARB_gpu_shader_int64     = not supported
ERROR: Link of shader progam failed:
Compute shader(s) failed to link.
Compute link error: HW_UNSUPPORTED.
ERROR: Internal compile error, error code: E_SC_NOTSUPPORTED
Shader not supported by HW
INTERNAL ERROR: failed to link histogram waveform shader program, aborting
    This indicates a bug in the program, please file a report via Github
`

Output of this utility: http://www.realtech-vr.com/home/glview:
`<?xml version="1.0" encoding="UTF-8" ?>

<GLView version="6.0.5"><System>Windows 10 x64 Edition  (Build 9200)</System>
<Vendor>ATI Technologies Inc.</Vendor>
<Version>4.6.14742 Core Profile Forward-Compatible Context 21.8.1 27.20.22023.1004</Version>
<GLSLVersion>4.60</GLSLVersion>
<DirectXVersion>DirectX 9, 12.1, 11.4</DirectXVersion>
<DXShaderModel>5.0, 4.1</DXShaderModel>
<Renderer>AMD Radeon RX 6800</Renderer>
<Extensions>GL_AMDX_debug_output GL_AMD_blend_minmax_factor GL_AMD_conservative_depth GL_AMD_debug_output GL_AMD_depth_clamp_separate GL_AMD_draw_buffers_blend GL_AMD_framebuffer_sample_positions GL_AMD_gcn_shader GL_AMD_gpu_shader_half_float GL_AMD_gpu_shader_half_float_fetch GL_AMD_gpu_shader_int16 GL_AMD_gpu_shader_int64 GL_AMD_interleaved_elements GL_AMD_multi_draw_indirect GL_AMD_name_gen_delete GL_AMD_occlusion_query_event GL_AMD_performance_monitor GL_AMD_pinned_memory GL_AMD_query_buffer_object GL_AMD_sample_positions GL_AMD_seamless_cubemap_per_texture GL_AMD_shader_atomic_counter_ops GL_AMD_shader_stencil_export GL_AMD_shader_stencil_value_export GL_AMD_shader_trace GL_AMD_shader_trinary_minmax GL_AMD_sparse_texture GL_AMD_sparse_texture_pool GL_AMD_stencil_operation_extended GL_AMD_texture_cube_map_array GL_AMD_texture_texture4 GL_AMD_transform_feedback3_lines_triangles GL_AMD_transform_feedback4 GL_AMD_vertex_shader_layer GL_AMD_vertex_shader_viewport_index GL_ARB_ES2_compatibility GL_ARB_ES3_1_compatibility GL_ARB_ES3_compatibility GL_ARB_arrays_of_arrays GL_ARB_base_instance GL_ARB_bindless_texture GL_ARB_blend_func_extended GL_ARB_buffer_storage GL_ARB_clear_buffer_object GL_ARB_clear_texture GL_ARB_clip_control GL_ARB_color_buffer_float GL_ARB_compressed_texture_pixel_storage GL_ARB_compute_shader GL_ARB_conditional_render_inverted GL_ARB_conservative_depth GL_ARB_copy_buffer GL_ARB_copy_image GL_ARB_cull_distance GL_ARB_debug_output GL_ARB_depth_buffer_float GL_ARB_depth_clamp GL_ARB_depth_texture GL_ARB_derivative_control GL_ARB_direct_state_access GL_ARB_draw_buffers GL_ARB_draw_buffers_blend GL_ARB_draw_elements_base_vertex GL_ARB_draw_indirect GL_ARB_draw_instanced GL_ARB_enhanced_layouts GL_ARB_explicit_attrib_location GL_ARB_explicit_uniform_location GL_ARB_fragment_coord_conventions GL_ARB_fragment_layer_viewport GL_ARB_fragment_program GL_ARB_fragment_program_shadow GL_ARB_fragment_shader GL_ARB_framebuffer_no_attachments GL_ARB_framebuffer_object GL_ARB_framebuffer_sRGB GL_ARB_geometry_shader4 GL_ARB_get_program_binary GL_ARB_get_texture_sub_image GL_ARB_gl_spirv GL_ARB_gpu_shader5 GL_ARB_gpu_shader_fp64 GL_ARB_half_float_pixel GL_ARB_half_float_vertex GL_ARB_imaging GL_ARB_indirect_parameters GL_ARB_instanced_arrays GL_ARB_internalformat_query GL_ARB_internalformat_query2 GL_ARB_invalidate_subdata GL_ARB_map_buffer_alignment GL_ARB_map_buffer_range GL_ARB_multi_bind GL_ARB_multi_draw_indirect GL_ARB_multisample GL_ARB_multitexture GL_ARB_occlusion_query GL_ARB_occlusion_query2 GL_ARB_parallel_shader_compile GL_ARB_pipeline_statistics_query GL_ARB_pixel_buffer_object GL_ARB_point_parameters GL_ARB_point_sprite GL_ARB_polygon_offset_clamp GL_ARB_program_interface_query GL_ARB_provoking_vertex GL_ARB_query_buffer_object GL_ARB_robust_buffer_access_behavior GL_ARB_sample_shading GL_ARB_sampler_objects GL_ARB_seamless_cube_map GL_ARB_seamless_cubemap_per_texture GL_ARB_separate_shader_objects GL_ARB_shader_atomic_counter_ops GL_ARB_shader_atomic_counters GL_ARB_shader_ballot GL_ARB_shader_bit_encoding GL_ARB_shader_draw_parameters GL_ARB_shader_group_vote GL_ARB_shader_image_load_store GL_ARB_shader_image_size GL_ARB_shader_objects GL_ARB_shader_precision GL_ARB_shader_stencil_export GL_ARB_shader_storage_buffer_object GL_ARB_shader_subroutine GL_ARB_shader_texture_image_samples GL_ARB_shader_texture_lod GL_ARB_shader_viewport_layer_array GL_ARB_shading_language_100 GL_ARB_shading_language_420pack GL_ARB_shading_language_include GL_ARB_shading_language_packing GL_ARB_shadow GL_ARB_shadow_ambient GL_ARB_sparse_buffer GL_ARB_sparse_texture GL_ARB_spirv_extensions GL_ARB_stencil_texturing GL_ARB_sync GL_ARB_tessellation_shader GL_ARB_texture_barrier GL_ARB_texture_border_clamp GL_ARB_texture_buffer_object GL_ARB_texture_buffer_object_rgb32 GL_ARB_texture_buffer_range GL_ARB_texture_compression GL_ARB_texture_compression_bptc GL_ARB_texture_compr [... cut ...] rent="False" acceleration="True" support_gdi="False" aux_buffers="0" double_buffer="True" support_opengl="True" draw_to_window="False" draw_to_bitmap="False" swap_layers_buffers="0" stereo="False" auxbuffers="0" samples="0" number_overlays="0" number_underlays="0" type_rgba_float="False" type_rgba_unsigned_float="False" framebuffer_srgb_capable="False"  /><p id="410" color_bits="32" depth_bits="0" stencil_bits="0" accum_bits="64" pixel_type="0" transparent="False" acceleration="True" support_gdi

ncicek, 2021-08-15:
Attached foo.txt
[foo.txt](https://github.com/azonenberg/scopehal-apps/files/6988698/foo.txt)

Attached output of glview utility.
[glview-AMD Radeon RX 6800-13.txt](https://github.com/azonenberg/scopehal-apps/files/6988699/glview-AMD.Radeon.RX.6800-13.txt)

TopQuark12, 2021-11-09:
I am facing a similar issue on a windows laptop with a AMD Ryzen 7 5800H processor.   
Here's my log:  
[log.txt](https://github.com/azonenberg/scopehal-apps/files/7502748/log.txt)

azonenberg, 2021-11-09:
Interesting. The log @ncicek posted looks similar to yours, and his glview dump confirms that GL_ARB_compute_shader is supported. Going to have to do some more digging...