ngscopeclient/scopehal-apps #387: Complie error on Windows X64, need help thanks!

TerryHoCQ, 2021-08-22:
$ cmake -G "MinGW Makefiles" -DLIBFFTS_INCLUDE_DIR=/mingw64/include/ \-DLIBFFTS_LIBRARIES=/mingw64/lib/libffts_static.a ./..
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/mingw64/bin/gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/msys64/mingw64/bin/g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: C:/msys64/mingw64/bin/pkg-config.exe (found version "1.7.4")
-- Checking for module 'gtkmm-3.0'
--   Found gtkmm-3.0, version 3.24.5
-- Checking for module 'sigc++-2.0'
--   Found sigc++-2.0, version 2.10.7
-- Checking for module 'ffts'
--   Package 'ffts', required by 'virtual:world', not found
-- Found libffts: C:/msys64/mingw64/include, C:/msys64/mingw64/lib/libffts_static.a
-- Looking for CL_VERSION_2_2
-- Looking for CL_VERSION_2_2 - not found
-- Looking for CL_VERSION_2_1
-- Looking for CL_VERSION_2_1 - not found
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - not found
-- Looking for CL_VERSION_1_2
-- Looking for CL_VERSION_1_2 - not found
-- Looking for CL_VERSION_1_1
-- Looking for CL_VERSION_1_1 - not found
-- Looking for CL_VERSION_1_0
-- Looking for CL_VERSION_1_0 - not found
-- Could NOT find OpenCL (missing: OpenCL_LIBRARY OpenCL_INCLUDE_DIR)
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Checking for one of the modules 'clFFT>=2.0'
-- Found GLEW: C:/msys64/mingw64/lib/cmake/glew/glew-config.cmake
-- Found OpenGL: opengl32
-- Found Git: C:/msys64/usr/bin/git.exe (found version "2.32.0")
Skipping MSI package build; define WIXPATH to enable
CMake Error at CMakeLists.txt:115 (find_package):
  By not providing "FindCatch2.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Catch2", but
  CMake did not find one.

  Could not find a package configuration file provided by "Catch2" with any
  of the following names:

    Catch2Config.cmake
    catch2-config.cmake

  Add the installation prefix of "Catch2" to CMAKE_PREFIX_PATH or set
  "Catch2_DIR" to a directory containing one of the above files.  If "Catch2"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!

azonenberg, 2021-08-22:
You're missing catch2, the unit testing framework we use for building test cases. Unless you plan to be writing test cases or working on the handful of features we have tests for so far (we're just starting to roll out unit testing), you can probably just turn it off by running `cmake -DBUILD_TESTING=OFF`.

TerryHoCQ, 2021-08-22:
Thanks, but error still exist.

$ mingw32-make -j4
[  0%] Building CXX object lib/log/CMakeFiles/log.dir/log.cpp.obj
[  1%] Building CXX object lib/xptools/CMakeFiles/xptools.dir/Socket.cpp.obj
[  2%] Building CXX object lib/graphwidget/CMakeFiles/graphwidget.dir/Graph.cpp.obj
[  2%] Copying styles...
[  2%] Built target styles
[  3%] Building CXX object lib/log/CMakeFiles/log.dir/ColoredSTDLogSink.cpp.obj
[  3%] Building CXX object lib/xptools/CMakeFiles/xptools.dir/UART.cpp.obj
[  3%] Building CXX object lib/log/CMakeFiles/log.dir/STDLogSink.cpp.obj
[  4%] Copying kernels...
[  4%] Built target kernels
[  5%] Building CXX object lib/log/CMakeFiles/log.dir/FILELogSink.cpp.obj
C:\msys64\home\Terry\scopehal-apps\lib\xptools\UART.cpp: In member function 'bool UART::Connect(const string&, int)':
C:\msys64\home\Terry\scopehal-apps\lib\xptools\UART.cpp:119:3: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
  119 |   (const void)devfile;
      |   ^~~~~~~~~~~~~~~~~~~
C:\msys64\home\Terry\scopehal-apps\lib\xptools\UART.cpp:120:3: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
  120 |   (const void)baud;
      |   ^~~~~~~~~~~~~~~~
[  5%] Copying icons...
[  6%] Linking CXX static library libxptools.a
[  6%] Built target icons
[  7%] Copying shaders...
[  7%] Built target xptools
[  7%] Built target shaders
[  7%] Linking CXX shared library liblog.dll
[  7%] Copying gradients...
[  7%] Built target gradients
[  7%] Linking CXX shared library libgraphwidget.dll
[  7%] Built target log
[  7%] Built target graphwidget
[  8%] Building CXX object lib/scopehal/CMakeFiles/scopehal.dir/base64.cpp.obj
[  8%] Building CXX object lib/scopehal/CMakeFiles/scopehal.dir/scopehal.cpp.obj
[  9%] Building CXX object lib/scopehal/CMakeFiles/scopehal.dir/avx_mathfun.cpp.obj
[  9%] Building CXX object lib/scopehal/CMakeFiles/scopehal.dir/Unit.cpp.obj
[ 10%] Building CXX object lib/scopehal/CMakeFiles/scopehal.dir/SCPITransport.cpp.obj
[ 10%] Building CXX object lib/scopehal/CMakeFiles/scopehal.dir/SCPISocketTransport.cpp.obj
In file included from C:\msys64\home\Terry\scopehal-apps\lib\scopehal\DemoOscilloscope.h:39,
                 from C:\msys64\home\Terry\scopehal-apps\lib\scopehal\scopehal.cpp:42:
C:\msys64\home\Terry\scopehal-apps\lib\scopehal\TestWaveformSource.h:40:10: fatal error: ffts.h: No such file or directory
   40 | #include <ffts.h>
      |          ^~~~~~~~
compilation terminated.
mingw32-make[2]: *** [lib\scopehal\CMakeFiles\scopehal.dir\build.make:91: lib/scopehal/CMakeFiles/scopehal.dir/scopehal.cpp.obj] Error 1
mingw32-make[2]: *** Waiting for unfinished jobs....
mingw32-make[1]: *** [CMakeFiles\Makefile2:241: lib/scopehal/CMakeFiles/scopehal.dir/all] Error 2
mingw32-make: *** [Makefile:135: all] Error 2

azonenberg, 2021-08-22:
Missing more dependencies. Install FFTS via your package manager (https://packages.msys2.org/package/mingw-w64-x86_64-ffts?repo=mingw64).

TerryHoCQ, 2021-08-22:
Hello, Confirmed the  FFTS  was installed. And I rebuild the src, the error as same as the above.

$ pacman -S mingw-w64-x86_64-ffts
warning: mingw-w64-x86_64-ffts-0.0.0.r799.gfe86885-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) mingw-w64-x86_64-ffts-0.0.0.r799.gfe86885-1

Total Installed Size:  0.18 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] y
(1/1) checking keys in keyring                     [#####################] 100%
(1/1) checking package integrity                   [#####################] 100%
(1/1) loading package files                        [#####################] 100%
(1/1) checking for file conflicts                  [#####################] 100%
(1/1) checking available disk space                [#####################] 100%
:: Processing package changes...
(1/1) reinstalling mingw-w64-x86_64-ffts           [#####################] 100%

TerryHoCQ, 2021-08-22:
Hello, copy "ffts.h" to "scopehal-apps/lib/scopehal" solved the error. And found another error as below:
[ 89%] Building CXX object src/glscopeclient/CMakeFiles/glscopeclient.dir/PreferenceTree.cpp.obj
[ 89%] Building CXX object src/glscopeclient/CMakeFiles/glscopeclient.dir/PreferenceManager.cpp.obj
mingw32-make[2]: *** [src\glscopeclient\CMakeFiles\glscopeclient.dir\build.make:256: src/glscopeclient/CMakeFiles/glscopeclient.dir/OscilloscopeWindow.cpp.obj] Error 1
mingw32-make[2]: *** Waiting for unfinished jobs....
In file included from C:/msys64/mingw64/x86_64-w64-mingw32/include/ws2tcpip.h:17,
                 from C:/msys64/home/Terry/scopehal-apps/lib/xptools/Socket.h:42,
                 from C:/msys64/home/Terry/scopehal-apps/lib/scopehal/SCPISocketTransport.h:39,
                 from C:/msys64/home/Terry/scopehal-apps/lib/scopehal/scopehal.h:76,
                 from C:\msys64\home\Terry\scopehal-apps\src\glscopeclient\glscopeclient.h:50,
                 from C:\msys64\home\Terry\scopehal-apps\src\glscopeclient\PreferenceManager.cpp:42:
C:/msys64/mingw64/x86_64-w64-mingw32/include/winsock2.h:15:2: warning: #warning Please include winsock2.h before windows.h [-Wcpp]
   15 | #warning Please include winsock2.h before windows.h
      |  ^~~~~~~
In file included from C:\msys64\home\Terry\scopehal-apps\src\glscopeclient\glscopeclient.h:74,
                 from C:\msys64\home\Terry\scopehal-apps\src\glscopeclient\PreferenceManager.cpp:42:
C:\msys64\home\Terry\scopehal-apps\src\glscopeclient\Program.h: In member function 'GLint Program::GetAttributeLocation(const char*)':
C:\msys64\home\Terry\scopehal-apps\src\glscopeclient\Program.h:64: note: '-Wmisleading-indentation' is disabled from this point onwards, since column-tracking was disabled due to the size of the code/headers
   64 |   GLint index = glGetAttribLocation(m_handle, name);
      |
mingw32-make[1]: *** [CMakeFiles\Makefile2:357: src/glscopeclient/CMakeFiles/glscopeclient.dir/all] Error 2

TerryHoCQ, 2021-08-22:
**2 more errors can not be solved, need help, thanks!**

[ 88%] Building CXX object src/glscopeclient/CMakeFiles/glscopeclient.dir/PreferenceDialog.cpp.obj
mingw32-make[2]: *** [src\glscopeclient\CMakeFiles\glscopeclient.dir\build.make:256: src/glscopeclient/CMakeFiles/glscopeclient
mingw32-make[2]: *** Waiting for unfinished jobs....
In file included from C:\msys64\home\Terry\scopehal-apps\src\glscopeclient\glscopeclient.h:74,
                 from C:\msys64\home\Terry\scopehal-apps\src\glscopeclient\PreferenceManager.cpp:43:
C:\msys64\home\Terry\scopehal-apps\src\glscopeclient\Program.h: In member function 'GLint Program::GetAttributeLocation(const c
C:\msys64\home\Terry\scopehal-apps\src\glscopeclient\Program.h:64: note: '-Wmisleading-indentation' is disabled from this pointhe size of the code/headers
   64 |   GLint index = glGetAttribLocation(m_handle, name);
      |
mingw32-make[1]: *** [CMakeFiles\Makefile2:357: src/glscopeclient/CMakeFiles/glscopeclient.dir/all] Error 2
mingw32-make: *** [Makefile:135: all] Error 2

closed 2022-10-03 by azonenberg