ngscopeclient/scopehal-apps #395: Implausible values in spectrum readout on TEK MSO68B

robot-army, 2021-12-18:
[image]

robot-army, 2021-12-18:
Oops, hit send too quickly! 

There are implausible values (-3601484742656 dB, etc) in the spectrum readouts on multiple channels being read from the 'scope. 
[image]

The screen on the scope looks like the attached (potatocam) photo.

robot-army, 2021-12-18:
running with --debug: 

```
$ ./glscopeclient.exe --debug
GetConsoleScreenBufferInfo() failed (6);
Warning: glscopeclient works best with the OMP_WAIT_POLICY environment variable
set to PASSIVE
Detecting CPU features...
    * AVX2
    * FMA

    Warning: AVX2/AVX512 detected but disabled on MinGW64/GCC (see https://githu
    b.com/azonenberg/scopehal-apps/issues/295)
OpenCL support: not present at compile time. GPU acceleration disabled.

Connecting to SCPI oscilloscope at 192.168.25.101:4000
Installed options:
* BW6-10000 (unknown)
* LIC6-DEMO (unknown)
Context: OpenGL 4.2 compatibility profile
    GL_VENDOR                   = NVIDIA Corporation
    GL_RENDERER                 = GeForce GTX 770/PCIe/SSE2
    GL_VERSION                  = 4.2.0 NVIDIA 376.51
    GL_SHADING_LANGUAGE_VERSION = 4.20 NVIDIA via Cg compiler
    Initial GL error code       = 0
    GL_ARB_gpu_shader_int64     = supported
Requested rendering backend: OpenGL
Using rendering backend: OpenGL
        Warning: Socket read failed (errno=0, No error)
        Warning: Timeout, attempting to recover

```

azonenberg, 2023-03-09:
There's a been a lot of work on the MSO6 driver over the past year or so, is this fixed?

robot-army, 2023-03-09:
Wish I could help... We sent the tek back!

azonenberg, 2023-03-09:
Closing for now, will reopen if anyone else has the same issue then.

closed 2023-03-09 by azonenberg