ngscopeclient/scopehal-apps #398: Fail to import touchstone from File

TerryHoCQ, 2021-12-25:
Merry Christmas! I'm trying to generate TDR waveform from a S2p file (On offline mode), and  I found 2 issues on importing touchstone file, thank you!

1: Fail to import touchstone from File menu
2: Can import touchstone file from Add/import menu, but no waveform generated.

azonenberg, 2021-12-25:
Please paste output of running glscopeclient with "--debug --trace TouchstoneParser" and trying to load your test file.

Also, are you able to share the .s2p file so I can try reproducing on my end?

TerryHoCQ, 2021-12-25:
Thank you! And please download the .s2p file from below link:
[http://www.sisolver.com/temp/SxP.zip](url)

azonenberg, 2021-12-25:
Testing on my end, all three files appear to load correctly.

There's an issue from the refactoring of the channel emulation / de-embed filter that makes it not work with touchstone files opened via file|import, you have to do add|import for it to recognize the file. This will be fixed at some point soon.

When you do add|import, do you not see a mag/angle waveform? Can you generate other test waveforms, say add|generate|sine, and see correct waveforms there? Are you testing on Linux or Windows?

[image]
[image]

TerryHoCQ, 2021-12-26:
Hello! I tested on Windows. Yes, add|import function works on test.s2p file, and failed on 12.s2p file. Here are the snapshots for your reference, thank you!
[image]
[image]

azonenberg, 2021-12-26:
All of your files load fine for me on Linux... but interestingly, 12.s2p has Windows-style \r\n line terminations while test.s2p has Unix-style \n line terminators. Can you try using dos2unix or similar to change the line endings and confirm that \r\n line endings trigger the bug while \n do not? If so, we're looking at a platform specific line ending handling problem.

Unrelated: you'll probably want to have the TDR filter have step start = -1V and end=0V if your S-parameters don't contain a DC component, since the resting (zero reflection) state of the channel emulation filter is going to be 0V not 1V in this case.

azonenberg, 2021-12-26:
Also, migrating this ticket to the scopehal repo since it appears to be a bug in the TouchstoneParser class in libscopehal, rather than in the glscopeclient UI.