ngscopeclient/scopehal-apps #399: Reload configuration from scope is tied to window events

mubes, 2022-01-13:
After clicking the 'Reload Config' option (third from right) on the toolbar, the reload 'steps' only happen when window events occur (e.g. mouse move). If there are no window events then the reload steps don't happen and things appear to freeze.

Tested on Siglent driver against 18e6d57fc4bf51198176cda765702853782cf79b.

DAVE

azonenberg, 2022-01-13:
The reload button flushes the configuration cache. Each individual configuration setting is pulled from the hardware the next time the relevant API, e.g. GetVoltageRange() is called. Some of these are called every frame while others are only called when you open the trigger or timebase properties dialog, the context menu, etc.

I guess maybe we could force a redraw of all waveform areas on the button press as well as flushing the cache?

mubes, 2022-01-13:
Hiya,

 From a UI perspective I think that would add a _lot_ of value, because 
the only reason you would flush the cache is that you think something is 
out of step in the display....it would also make it 'feel' like it's 
done something (and make this bug I'm trying to find a lot easier to 
track down!).

Regards

DAVE

azonenberg, 2022-01-13:
I don't have time to work on this for a bit but try adding a queue_draw() call in the refresh handler in OscilloscopeWindow. You might have to also loop over the WaveformArea instances and refresh them too, I forget if it's recursive to child windows or not.

mubes, 2022-01-14:
Candidate patch available as https://github.com/azonenberg/scopehal-apps/pull/400

DAVE

azonenberg, 2023-03-21:
Fixed in 1ed6e626bde5e60bd2ae769bd96b8355baa2d12d

closed 2023-03-21 by azonenberg