ngscopeclient/scopehal-apps #433: View color ramp setting is not saved / restored in *.scopesession

bvernoux, 2022-06-17:


azonenberg, 2022-06-17:
The current color ramp was intended to be a user preference, not a file setting (i.e. it should follow the user and if you load a scopesession on another user's workstation it should use their preferred colors).

It's not currently saved as such. We should have a greater discussion as to whether it makes sense for this to be a per-file or per-user setting.

Also, I'd like to be able to have more than one ramp selected for different uses (e.g. eye patterns and spectrograms might not be best to use the same ramp).

bvernoux, 2022-06-17:
Yes that feature clearly requires refactoring to be stored with scopesession per chan requiring it
Short term could be to have a way to change globally the default "View->Color ramp" to something different from default "CRT" (as it is clearly ugly)

azonenberg, 2023-11-03:
Fixed in ngscopeclient from day one, glscopeclient is deprecated so closing this

closed 2023-11-03 by azonenberg