ngscopeclient/scopehal-apps #435: Marker issue

bvernoux, 2022-06-17:
When using Marker all work fine visually but when saving/restoring a *.scopesession the markers are broken with invalid values(negatives) and when the scopesession is loaded nothing is displayed (even the waveform).
Issue reproduced with latest version of glscopeclient https://github.com/glscopeclient/scopehal-apps/commit/af2c022e85fbb9905960ebc272c86d3319ccc196 on windows 10 (using https://github.com/glscopeclient/scopehal-apps/suites/6973742607/artifacts/272626580)
File used are based on Rigol MSO5K import (with 2 chan) I can provide them if required

azonenberg, 2022-06-17:
Please attach test files and screenshots.

azonenberg, 2022-06-17:
Also, please check if the "nothing displayed on waveform" issue is related to https://github.com/glscopeclient/scopehal-apps/commit/524c645edb20eccabaa7e463b915af3647d72aea.

Does reverting to prior to this commit fix it?

bvernoux, 2022-06-17:
It seems this issue is also affected by the new bug https://github.com/glscopeclient/scopehal-apps/issues/436 (newly introduced in latest fix ?? as I have never seen that in older glscopeclient used 1 week ago), so I will wait this bug to be fixed before

azonenberg, 2022-06-17:
Please check if f1d109bbd4c8d4d3553beab9bfa08da5f1638c5d fixes the rendering issue.

bvernoux, 2022-06-19:
This issue is related to https://github.com/glscopeclient/scopehal-apps/issues/436

azonenberg, 2022-06-27:
Closing this and tracking #442 as a separate issue for "markers don't work in sessions without an attached scope".

closed 2022-06-27 by azonenberg