ngscopeclient/scopehal-apps #436: Rigol MSO5K bin import issue after save/load scopesession

bvernoux, 2022-06-17:
Rigol MSO5K bin import issue after save/load scopesession
See full steps with files to reproduce the issue
1) Import the file **CH569_SerDes_Speed_IRQ_inline_optim_Chan1_TX_Chan2_RX_5us.bin** from archive
[CH569_SerDes_Speed_IRQ_inline_optim_Chan1_TX_Chan2_RX_5us.zip](https://github.com/glscopeclient/scopehal-apps/files/8928813/CH569_SerDes_Speed_IRQ_inline_optim_Chan1_TX_Chan2_RX_5us.zip)
2) All is fine and perfectly displayed (need zoom out fully to see the full capture)
* Minor issue 2 channels does not have any name it shall be something like Chan1, Chan2
[image]
3) Click on File => Save Layout and Waveforms As 
- Save the file in easy path like D:\Test (so both files Test.scopesession & Test_data directory with sub-dir/files are created with success)
- See the archive [Test.zip](https://github.com/glscopeclient/scopehal-apps/files/8928867/Test.zip)
4) Choose File => Close
5) Choose File => Open... and choose  D:\Test.scopesession
I cannot zoom out to see the previous screen like on 2)
[image]

bvernoux, 2022-06-18:
This issue does not appear when using glscopeclient version 9e8cdce (https://github.com/glscopeclient/scopehal-apps/commit/9e8cdcef3604eaee5fd1b7b2cd5d3dfe5d856759) (see https://github.com/glscopeclient/scopehal-apps/actions/runs/2442228348)

azonenberg, 2022-06-27:
Please test if fixed by f69e6d50afd9f6c7d9ce8ae107a36636f167ac9b

bvernoux, 2022-06-27:
The Import work but when we save the session and we try to load it again glscopeclient quit (like a crash without any details) ...
You can reproduce that with the file provided doing all the steps in my first comment

azonenberg, 2022-06-27:
Yes, there is no longer any history involved as the new import flow is filter based, and filters don't have any concept of history.

I just tried in 64c72799596eeb0e2b4f67ca444e1d36cd78a96e on Linux. Imported your rx_5us file, saved to a scopesession, then opened it. Loaded correctly, no crash, waveforms look fine.

Can you try building from source with debug symbols and running under a debugger to see if you can narrow down what's going on? Do you have access to a suitable Linux dev machine you can try on?

closed 2022-06-27 by bvernoux