ngscopeclient/scopehal-apps #437: Add a way to truncate waveform(s)

bvernoux, 2022-06-17:
Add a way to truncate waveform(s) with start and end markers for example to truncate and keep only the part between markers
This will be applied on all waveform(s) if there is several waveforms displayed

azonenberg, 2022-10-23:
@602p can we close this now that you have the clip filter?

602p, 2022-10-23:
Not sure this ask is fully met since the window filter still stores the entire waveform in the history etc. But it's largely solved.

bvernoux, 2022-10-23:
So far Clip (tested in nglscopeclient) does not allow to cut a signal (left & right par for example) but clip on top/bottom

602p, 2022-10-23:
@bvernoux I believe azonenberg made a typo; the 'Window' filter allows selecting a temporal subset of a waveform

closed 2023-09-29 by azonenberg