ngscopeclient/scopehal-apps #460: Windows recentfiles.yml YAML::ParserException at startup

bvernoux, 2022-07-31:
Windows recentfiles.yml YAML::ParserException at startup (even with latest build)
Return error
terminate called after throwing an instance of 'YAML::ParserException'
  what():  yaml-cpp: error at line 2, column 42: bad character found while scanning hex number

Path C:\Users\xxx\AppData\Roaming\glscopeclient\recentfiles.yml is not formatted correctly (I suspect it is because of path)
See content of the file
```
file0:
    path: "D:\_proj\__Lab_Tools\USB3_USB2HS_Gigabit\CH569\wch-ch569-serdes\CH1_TXSTATE_Ch2_RXIRQ_UART_CH4_SerDes_180Mbps\CH1_TXSTATE_Ch2_RXIRQ_UART_CH4_SerDes_180Mbps.bin.scopesession"
    timestamp: 1659296109
```

Solution until it is fixed is to remove that recentfiles.yml before to launch glscopeclient

azonenberg, 2022-08-03:
Can you take a look at OscilloscopeWindow.cpp:4768 and see if escpath is properly escaped? The code *looks* fine to me but I don't have a Windows box to test on.

bvernoux, 2022-08-07:
The issue seems related to length of path+file
It is not really clear so far as with shorter path it works

bvernoux, 2023-06-09:
Issue closed as it cannot be reproduced and was not related to scopehal-apps