ngscopeclient/scopehal-apps #470: Windows CI build is broken despite finding Vulkan SDK

azonenberg, 2022-08-18:
[code]
 -- Could NOT find Vulkan (missing: Vulkan_LIBRARY) (found version "1.3.223")
CMake Error at CMakeLists.txt:61 (message):
  No Vulkan SDK found.
[/code]

stanciuadrian, 2022-08-18:
It might not work because the way environment variables are inherited in the CI.

Let's say process `A` is the CI runner and during its execution it spawns process `B` which is the Vulkan SDK installer. `B` ends by successfully setting global environment variables. Then `A` spawns the build step as a new process `C`. `C` doesn't see the environment variables set by `B` because it inherits the ones from `A`.

My suggestion here is to:

1. force the installation path with the `--root` command line argument while installing the Vulkan SDK
2. [set](https://docs.github.com/en/actions/learn-github-actions/environment-variables) the `VK_SDK_PATH` and `VULKAN_SDK` to the same path in the MSYS2 build step.

stanciuadrian, 2022-08-18:
You may find [act](https://github.com/nektos/act) useful for quickly iterating the CI changes.

I can't make Docker work on my machine, otherwise I would have helped you more with it.

Johnsel, 2022-08-19:
I have been able to reproduce this same issue on a local Windows install so I can work on figuring this out tomorrow.

Biswa96, 2022-08-31:
Why was [mingw-w64-vulkan-loader](https://packages.msys2.org/base/mingw-w64-vulkan-loader) package not being used?

Biswa96, 2022-08-31:
This workaround may fix the Windows CI build.

```diff
--- a/msys2/PKGBUILD
+++ b/msys2/PKGBUILD
@@ -48,6 +48,7 @@ build() {
     -G "Ninja" \
     -DCMAKE_INSTALL_PREFIX="${MINGW_PREFIX}" \
     -DBUILD_TESTING=OFF \
+    -DVulkan_LIBRARY=${MINGW_PREFIX}/lib/libvulkan.dll.a \
     ../
 
   ${MINGW_PREFIX}/bin/cmake.exe --build ./
```

But this is a workaround till now. The `libvulkan.dll.a` name may change to `libvulkan-1.dll.a` in future. There is some debate over what the vulkan DLL and import library should be in mingw environment. Maybe I started that controversy 🙇 Feel free to ping me anytime anywhere.

azonenberg, 2022-08-31:
(paging @bvernoux since he's working on this)

bvernoux, 2022-08-31:
I have found a fix see https://github.com/bvernoux/scopehal-apps/blob/master/.github/workflows/build.yml
See the build here https://github.com/bvernoux/scopehal-apps/runs/8120159932
I have a last issue is the path to glslc is not found even if I have exported it before 
`export PATH=/c/VulkanSDK/1.3.224.1/Bin:$PATH`
https://github.com/bvernoux/scopehal-apps/runs/8120159932?check_suite_focus=true#step:6:1431
'glslc' is not recognized as an internal or external command,
We shall find a way to set this path in ninja instance env
Note: When I build manually with mingw32-make all work fine on my computer and I can build glscopeclient.exe with success

Biswa96, 2022-08-31:
That's OK. The solution is a bit different. The executables may link with the MSVC compiled vulkan libraries. Setting `VULKAN_SDK` will satisfy this condition in cmake's FindVulkan module.

```cmake
    set(_Vulkan_hint_library_search_paths
      "$ENV{VULKAN_SDK}/Lib"
      "$ENV{VULKAN_SDK}/Bin"
    )
```

Whereas, setting `Vulkan_LIBRARY` will satisfy this condition

```cmake
find_library(Vulkan_LIBRARY
  NAMES ${_Vulkan_library_name}
  HINTS
    ${_Vulkan_hint_library_search_paths}
  )
mark_as_advanced(Vulkan_LIBRARY)
```

But both reach to same destination.

bvernoux, 2022-08-31:
> It might not work because of the way environment variables are inherited in the CI.
> 
> Let's say process `A` is the CI runner and during its execution it spawns process `B` which is the Vulkan SDK installer. `B` ends by successfully setting global environment variables. Then `A` spawns the build step as a new process `C`. `C` doesn't see the environment variables set by `B` because it inherits the ones from `A`.
> 
> My suggestion here is to:
> 
>     1. force the installation path with the `--root` command line argument while installing the Vulkan SDK
> 
>     2. [set](https://docs.github.com/en/actions/learn-github-actions/environment-variables) the `VK_SDK_PATH` and `VULKAN_SDK` to the same path in the MSYS2 build step.

Yes there is clearly something like that which is broken and shall be fixed

bvernoux, 2022-09-01:
Build fixed see https://github.com/glscopeclient/scopehal-apps/pull/481

Johnsel, 2022-09-02:
@bvernoux on https://github.com/Johnsel/scopehal-apps/runs/8164338202 glscopeclient --help runs too now. I'm not entirely sure what has fixed it, presumably something related to the errors or the shell. Perhaps you will be able to figure it out with fresh eyes on it.

bvernoux, 2022-09-04:
@Johnsel Could you create a PR with your modifications ? (that is really better than what I have done and that seems to fix issues with msys2 shell with explicit definition which is always better ...)

Johnsel, 2022-09-05:
I could, but I am still confused about the build test phase fixing itself with seemingly unrelated changes so I'd like to confirm that we have a fully functional build before we switch over. Is the glslc issue fixed? The missing header I mean.

bvernoux, 2022-09-07:
glslc is not fixed as the required headers are not present in Windows VulkanSDK but present for Linux (an issue shall be created for VulkanSDK to ask they add those headers and static lib if some are missing)
The fix for Linux is easy and I have it working (requires to add 2 lines in the Cmake)
I'm searching a way to also fix the windows build but so far  it will requires to clone glslc git and rebuild it

azonenberg, 2022-09-07:
Is it possible that on Windows we might have to install https://github.com/KhronosGroup/glslang/ separately from the SDK?

bvernoux, 2022-09-07:
> Is it possible that on Windows we might have to install https://github.com/KhronosGroup/glslang/ separately from the SDK?

Yes so far it seems to be the only way to do it until VulkanSDK add it (maybe that also impact OSX to be checked)

Johnsel, 2022-09-07:
Do we need to find this header in the VulkanSDK (or glslang) though? It seems that the mingw package that supplies glslc should provide this, no? For some reason it does not properly add that path to the -I includes though. I can work on it some more tomorrow,

Johnsel, 2022-09-07:
https://packages.msys2.org/package/mingw-w64-x86_64-glslang this one I mean

Johnsel, 2022-09-07:
See one of the first 10-ish includes at https://github.com/glscopeclient/scopehal-apps/runs/8210668826?check_suite_focus=true#step:7:257 it is refering to /usr instead of the MinGW prefixed path

bvernoux, 2022-09-07:
> https://packages.msys2.org/package/mingw-w64-x86_64-glslang this one I mean

Yes it is an other easier (potentially better) solution
I have not fully checked that especially to check if all lib are ok for the link

Biswa96, 2022-09-09:
Update. The mingw-w64 vulkan loader package has been fixed in msys2. [The workaround][1] is not required anymore.

[1]: https://github.com/glscopeclient/scopehal-apps/issues/470#issuecomment-1233332518

Biswa96, 2022-09-09:
Update. The mingw-w64 vulkan loader package has been fixed in msys2. [The workaround][1] is not required anymore.

[1]: https://github.com/glscopeclient/scopehal-apps/issues/470#issuecomment-1233315591

bvernoux, 2022-09-10:
> Update. The mingw-w64 vulkan loader package has been fixed in msys2. [The workaround](https://github.com/glscopeclient/scopehal-apps/issues/470#issuecomment-1233315591) is not required anymore.

I do not understand what is your workaround as we are using VulkanSDK and so we shall use VulkanSDK lib
Anyway so far the actual master have different issues to be fixed

- I was testing to integrate [mingw-w64-x86_64-glslang](https://packages.msys2.org/package/mingw-w64-x86_64-glslang) but it does not work so the best seems to be using https://github.com/KhronosGroup/glslang.git and rebuild it for MINGW64
- Short/Mid/Long Term is waiting a fix/new version of VulkanSDK for Windows see issue https://vulkan.lunarg.com/issue/view/631adb345df112f88f25d523

closed 2022-09-11 by azonenberg