ngscopeclient/scopehal-apps #471: Look at alternative CI platforms

azonenberg, 2022-08-18:
GitHub Actions was great for getting started but doesnt seem to support ARM64 Apple platforms (among others). We might want to move to something more general (CDash? We're already using CTest so it would be easy integration)

Johnsel, 2022-08-19:
It is very much possible to add other platform support by providing your own runners and docker containers to GitHub Actions https://docs.github.com/en/actions/hosting-your-own-runners/about-self-hosted-runners

There is a lot of useful integrations that you might lose or which may become time-intensive to manage if you go for another solution. You would also most likely still end up setting up your own runners and containers on whatever other platform you choose.

azonenberg, 2022-08-19:
According to the docs, self hosted runners are not supported for any platform but Linux on the ARM64 ISA.

It's not a question of self hosting vs third party, it's a question of whether CI for our Mac port is even *possible* on GitHub Actions or not.

Johnsel, 2022-08-31:
Just so it's 'on paper'. It is perfectly possible to set up a mac m1 runner within the GitHub CI system, but the Windows end of things has proven far more frustrating as errors end up in the Windows equivalent of /dev/null. This may be due to msys2/mingw, or due to some other reason why errors end up in e.g. a messagebox instead of console. Simultaneous GL initialization? Something else entirely? I was not able to figure it out in the time I allotted to the problem.

azonenberg, 2024-01-29:
Closing as I think we have a good path forward on this now.

closed 2024-01-29 by azonenberg