ngscopeclient/scopehal-apps #475: Trigger properties update for every digit entered

dan-gies, 2022-08-22:
In the trigger properties dialog, the trigger updates live on the scope as I enter it into the "Level" field, instead of on close (or, preferably, on <ENTER> or "OK" button, if one existed).

Example: I want to enter 150mV trigger. I type "1", trigger is set to 1V on scope. I then type "5", trigger is set to 15V, which is outside of trigger limit. I type "0" then "m" and trigger is correctly set.

I don't think this poses a risk to the instrument, but it isn't how I expected the UI to function.

azonenberg, 2022-08-22:
We're trying to move to a more modeless/nonblocking UI flow so that changes occur live rather than waiting for you to close the dialog (so you can leave it open and tweak trigger settings, seeing the scope update in real time).

Maybe it would make more sense to trigger the push when you shift focus outside the text box rather than immediately on key presses? Open to discussion.

dan-gies, 2022-08-28:
I like the idea of triggering the push when you shift focus to a different text field, or on hitting `ENTER`.