ngscopeclient/scopehal-apps #478: Can't compile glscopeclient with Ubuntu 20.04 Fossa

minecraft2048, 2022-08-26:
Followed the instruction line by line at the getting started manual at here: https://www.antikernel.net/temp/glscopeclient-manual.pdf
with master branch commit 3bc6b5f29650c555a83cbe3dfbb39e3d00467496, when I tried to compile it it failed with missing glslc error:
```
n310@n310-alienware:~/scopehal-apps/build$ make -j8
[  1%] Building CXX object lib/log/CMakeFiles/log.dir/log.cpp.o
[  1%] Copying rendering kernels...
[  1%] Building CXX object lib/xptools/CMakeFiles/xptools.dir/Socket.cpp.o
[  2%] Copying icons...
[  2%] Copying gradients...
[  2%] Copying shaders...
[  2%] Compile shader Convert16BitSamples
/bin/sh: 1: glslc: not found
make[2]: *** [lib/scopeprotocols/shaders/CMakeFiles/protocolshaders.dir/build.make:61: lib/scopeprotocols/shaders/Convert16BitSamples.spv] Error 127
make[1]: *** [CMakeFiles/Makefile2:511: lib/scopeprotocols/shaders/CMakeFiles/protocolshaders.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  2%] Built target renderkernels
[  2%] Building CXX object lib/xptools/CMakeFiles/xptools.dir/UART.cpp.o
[  2%] Built target gradients
[  2%] Built target shaders
[  2%] Built target icons
[  2%] Building CXX object lib/log/CMakeFiles/log.dir/ColoredSTDLogSink.cpp.o
[  2%] Building CXX object lib/graphwidget/CMakeFiles/graphwidget.dir/Graph.cpp.o
[  2%] Building CXX object lib/log/CMakeFiles/log.dir/STDLogSink.cpp.o
[  3%] Building CXX object lib/log/CMakeFiles/log.dir/FILELogSink.cpp.o
[  4%] Linking CXX static library libxptools.a
[  4%] Built target xptools
[  4%] Linking CXX static library liblog.a
[  4%] Built target log
[  4%] Linking CXX shared library libgraphwidget.so
[  4%] Built target graphwidget
make: *** [Makefile:130: all] Error 2
```

azonenberg, 2022-08-26:
You need to install the Vulkan SDK from https://vulkan.lunarg.com/sdk/home#linux.

This dependency was added very recently and hasn't made it into the docs yet. If you have any problems you can roll back to 4a96248c588ed5c14818201c6135b14a6d385a04 and it should build fine without Vulkan.

minecraft2048, 2022-08-26:
Installed Vulkan SDK, compiled it and it crashes with this error:
```
n310@n310-alienware:~/scopehal-apps/build$ glscopeclient 
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

Warning: ReadDataFile: Could not open file "shaders/SubtractFilter.spv"
terminate called after throwing an instance of 'vk::UnknownError'
  what():  vkCreateShaderModule: ErrorUnknown
Aborted (core dumped)
```

minecraft2048, 2022-08-26:
If I try to revert to https://github.com/glscopeclient/scopehal-apps/commit/4a96248c588ed5c14818201c6135b14a6d385a04 CMake cannot find yaml config file:
```
n310@n310-alienware:~/scopehal-apps/build$ cmake ..
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gtkmm-3.0'
--   Found gtkmm-3.0, version 3.24.2
-- Checking for module 'sigc++-2.0'
--   Found sigc++-2.0, version 2.10.2
-- Checking for module 'ffts'
--   Variable 'exec_prefix' not defined in '/usr/local/lib/pkgconfig/ffts.pc'
-- Found libffts: /usr/local/include/ffts, /usr/local/lib/libffts.so
-- Looking for CL_VERSION_2_2
-- Looking for CL_VERSION_2_2 - found
-- Found OpenCL: /usr/lib/x86_64-linux-gnu/libOpenCL.so (found version "2.2") 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Checking for one of the modules 'clFFT>=2.0'
CMake Error at lib/scopehal/CMakeLists.txt:17 (find_package):
  By not providing "FindYAML.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "YAML", but
  CMake did not find one.

  Could not find a package configuration file provided by "YAML" with any of
  the following names:

    YAMLConfig.cmake
    yaml-config.cmake

  Add the installation prefix of "YAML" to CMAKE_PREFIX_PATH or set
  "YAML_DIR" to a directory containing one of the above files.  If "YAML"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/n310/scopehal-apps/build/CMakeFiles/CMakeOutput.log".
n310@n310-alienware:~/scopehal-apps/build$ 
```
while I can compile at master

electroniceel, 2022-08-26:
> ```
> Warning: ReadDataFile: Could not open file "shaders/SubtractFilter.spv"
> terminate called after throwing an instance of 'vk::UnknownError'
>   what():  vkCreateShaderModule: ErrorUnknown
> Aborted (core dumped)
> ```

This is a known issue. You can either copy the .spv files manually after the build or apply this patch: https://github.com/glscopeclient/scopehal/pull/671
The PR will probably be merged by @azonenberg when the current refactoring is done.

azonenberg, 2022-08-30:
Should be fixed. Reopen if you still see the issue now that everything is merged.

closed 2022-08-30 by azonenberg