ngscopeclient/scopehal-apps #480: Channel 1 waveform in glscopeclient differs from oscilloscope screen, timebase settings differ

dan-gies, 2022-08-28:
## System Info:
Glscopeclient version: 0.1-0ec5f20
OS: Windows 10
Oscilloscope: Siglent SDS6204A
Probe: SAP2500D

## Description:
The waveform I see on the oscilloscope is different from the waveform in glscopeclient. The timebase settings also differ between glscopeclient and the oscilloscope; glscopeclient won't allow me to configure the timebase settings that I see on the scope.

### glscopeclient waveform:
[image]

### Oscilloscope waveform:
[image]

### glscopeclient timebase settings (set to maximum possible sample rate, memory depth):
[image]

### Oscilloscope timebase settings:
[image]

dan-gies, 2022-08-28:
Reducing the memory depth manually (on the scope) to 12.5 Mpts worked. My assumption is that glscopeclient doesn't support 500 Mpts memory depth per channel.

[image]

azonenberg, 2023-03-09:
Max memory depth in the Siglent driver is I think 12.5M; the waveform download on Siglent scopes is incredibly slow with deep memory so we add an early-out to avoid timeouts on these huge waveforms.

+ @mubes, can we improve failure behavior here in the driver if the user has deep memory enabled when connecting to a scope where memory depth is higher than the driver allows?