ngscopeclient/scopehal-apps #507: Link error with reflowmon

RX14, 2022-09-28:
When running `cmake` I get the following error. Seems the correct link libraries aren't being passed to reflowmon when linking with libscopehal.

```
[ 38%] Linking CXX executable reflowmon
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::CreateCompactIdsPass()'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::Optimizer::PassToken::~PassToken()'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::Optimizer::RegisterPerformancePasses()'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvValidatorOptionsDestroy'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvContextDestroy'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvValidatorOptionsSetBeforeHlslLegalization'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::SpirvTools::SpirvTools(spv_target_env)'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::Optimizer::RegisterLegalizationPasses()'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvTextToBinary'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::Optimizer::Run(unsigned int const*, unsigned long, std::vector<unsigned int, std::allocator<unsigned int> >*, spv_optimizer_options_t*) const'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvDiagnosticDestroy'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvValidatorOptionsSetSkipBlockLayout'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::Optimizer::SetMessageConsumer(std::function<void (spv_message_level_t, char const*, spv_position_t const&, char const*)>)'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::Optimizer::~Optimizer()'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvOptimizerOptionsSetValidatorOptions'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvContextCreate'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::SpirvTools::SetMessageConsumer(std::function<void (spv_message_level_t, char const*, spv_position_t const&, char const*)>)'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvValidatorOptionsCreate'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::Optimizer::RegisterSizePasses()'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvOptimizerOptionsCreate'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::SpirvTools::Disassemble(std::vector<unsigned int, std::allocator<unsigned int> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, unsigned int) const'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvOptimizerOptionsSetRunValidator'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvOptimizerOptionsDestroy'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::Optimizer::Optimizer(spv_target_env)'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::SpirvTools::~SpirvTools()'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::CreateStripDebugInfoPass()'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvValidatorOptionsSetRelaxLogicalPointer'
/usr/bin/ld: ../../lib/scopehal/libscopehal.so: undefined reference to `spvtools::Optimizer::RegisterPass(spvtools::Optimizer::PassToken&&)'
collect2: error: ld returned 1 exit status
make[2]: *** [src/reflowmon/CMakeFiles/reflowmon.dir/build.make:133: src/reflowmon/reflowmon] Error 1
make[1]: *** [CMakeFiles/Makefile2:1797: src/reflowmon/CMakeFiles/reflowmon.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

azonenberg, 2022-09-28:
Reflowmon is an old legacy datalogging application that is entirely obsoleted by the multimeter history graph features in ngscopeclient. I'll delete it.

closed 2022-09-28 by azonenberg

RX14, 2022-10-04:
This is not just an error with `reflowmon`, it just happens to be linked first. The error occurs also when linking glscopeclient and ngscopeclient executables. Latest `master`.

meerfrau, 2023-10-12:
@RX14 Have you found a solution? I'm getting a similar error with a different program reported at https://github.com/google/shaderc/issues/1374

Johnsel, 2023-10-12:
@meerfrau are trying to compile ngscopeclient? If so please provide more details.

meerfrau, 2023-10-12:
@Johnsel No, but I similarly do lack reference to https://github.com/KhronosGroup/SPIRV-Tools symbols

Johnsel, 2023-10-12:
If your issue has nothing to do with our project you should not be posting about it here. Thanks for understanding.

commit 97dc33c6 (2022-09-28): Deleted old legacy reflowmon app. Fixes #507.