ngscopeclient/scopehal-apps #542: Feature request: highlight waveform span when using protocol analyzer

dan-gies, 2022-11-07:
Apologies if this isn't the right place to put feature requests.

I'm using the protocol analyzer, and find it incredibly useful to click on different packets in the protocol analyzer to jump to the timestamp in the waveform. However, when the waveform is zoomed out, it can be tough to tell what the highlighted (grey) packet refers to on the waveform. It would be great to have a same-colored bar on the time axis to indicate where in the waveform the packet refers:

[image]

azonenberg, 2022-11-07:
If you have an X axis cursor enabled, it should jump to the start of the packet when you select one.

dan-gies, 2022-11-07:
This works well! Thanks.

closed 2022-11-07 by dan-gies