ngscopeclient/scopehal-apps #561: Compilation error on Ubuntu 20.04

robbederks, 2022-12-04:
I've followed the "getting started" section of the manual pretty closely, but unfortunately I get a compilation error while trying to build scopehal-apps:

```
[  1%] Copying shaders...
[  2%] Built target xptools
[  2%] Copying styles...
[  2%] Copying icons...
[  3%] Built target log
[  3%] Copying icons...
[  6%] Built target protocolshaders
[  6%] Copying fonts...
[ 10%] Built target ngrendershaders
[ 10%] Copying gradients...
[ 10%] Built target styles
[ 10%] Built target shaders
[ 13%] Built target rendershaders
[ 14%] Built target ngcomputeshaders
[ 14%] Built target icons
[ 14%] Built target nfd
[ 14%] Built target ngfonts
[ 14%] Built target ngicons
[ 14%] Copying protocol shaders...
[ 14%] Built target gradients
Error copying file (if different) from "/home/robbe/glscopeclient/scopehal-apps/build/lib/scopeprotocols/shaders/CMakeFiles/protocolshaders" to "/home/robbe/glscopeclient/scopehal-apps/build/src/ngscopeclient/shaders".
make[2]: *** [src/ngscopeclient/CMakeFiles/ngprotoshaders.dir/build.make:58: src/ngscopeclient/CMakeFiles/ngprotoshaders] Error 1
make[1]: *** [CMakeFiles/Makefile2:1652: src/ngscopeclient/CMakeFiles/ngprotoshaders.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 15%] Built target graphwidget
[ 34%] Built target scopehal
make: *** [Makefile:141: all] Error 2
```

I've tried to revert the #539 PR, but that did not fix it for me.
I'll try to figure out what goes wrong and will update this issue with my findings!

P.S.: Another (small) note I have is that it's impossible to just copy-paste the command line instructions from the PDF (chrome PDF reader -> zsh). For some reason, it seems to add spaces before and after every special character. Maybe it would be nice to have some bash script to do the bulk of the setup work instead?

robbederks, 2022-12-04:
From the looks of it the .spv files were already in the destination folder somehow, so I just commented out the copy commands from the CMake files, and that seemed to have worked.

No idea how it came up with the (wrong) source path for them though, there's nothing there...

bvernoux, 2023-06-07:
The cmake from Ubuntu 20.04 is probably too old (buggy) as there is not such issue with Ubuntu 22.04 LTS or more

robbederks, 2023-06-07:
> The cmake from Ubuntu 20.04 is probably too old (buggy) as there is not such issue with Ubuntu 22.04 LTS or more
> Could you try again with your Ubuntu 20.04 on master ?

I have since upgraded to 22.04, but also haven't tried glscopeclient anymore.

Feel free to close the issue in the meantime, I'll re-open should I hit it again in the future!

azonenberg, 2023-06-13:
I think this was a race condition between the builds of glscopeclient and ngscopeclient that I fixed a few months back. Closing, reopen if it comes back.

closed 2023-06-13 by azonenberg