ngscopeclient/scopehal-apps #565: ngscopeclient: crash on macOS when attempting to use dropdown menu.

sparrowgrine, 2022-12-29:
ngscopeclient crashes with the error message `*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'` when run on macOS. 
[ngscopeclienttrace.txt](https://github.com/glscopeclient/scopehal-apps/files/10322330/ngscopeclienttrace.txt)

mandl, 2023-01-15:
Same on my machine. Using _"Open Online"_ "_Open Offline_"  _"Save as"_  ngscopeclient will crash.

**Background**:

For those operations we use **nativefiledialog-extended**
https://github.com/btzy/nativefiledialog-extended/

**Problem**: 

NFDFileBrowser creates a new thread
https://github.com/glscopeclient/scopehal-apps/blob/d48c7411061f7cfbec6888638f3254e1bd9e4ba3/src/ngscopeclient/NFDFileBrowser.cpp#L64

and later NFD_SaveDialog throw an exception. Can only render UI from main thread.

uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'

https://github.com/glscopeclient/scopehal-apps/blob/d48c7411061f7cfbec6888638f3254e1bd9e4ba3/src/ngscopeclient/NFDFileBrowser.cpp#L129

azonenberg, 2023-01-15:
First off, the file load/save functions do not actually work in ngscopeclient yet (they show a browser dialog then a warning saying the functionality is incomplete). So a partial workaround is to not click on them!

But this will also crash import filters for loading a VCD etc, since those spawn file browser dialogs.

You can work around this by selecting the non-fullscreened dialog style to ImGui under Appearance / File Browser in the preferences dialog. This will use IGFDFileBrowser instead of NFDFileBrowser. It won't have the proper MacOS look/feel, but will function correctly.

@mandl: want to send a pull request with a temporary workaround that forces use of IGFDFileBrowser instead of NFDFileBrowser on MacOS?

I'm not sure what the proper long term fix is, because our file browser wants to be nonblocking and NativeFileDialog functions re blocking. Reworking the code to handle a blocking dialog there would be awkward, and break a lot of things.

Also + @lainy since she's done a lot of the Mac portability work.