ngscopeclient/scopehal-apps #567: Fails to compile on Windows for me

Teddyz, 2023-01-06:
Compiling fails for me.
Happy if anyone could help me with this (or if it is just that I run Win7)
Vulkan cube works. It's configuration says Vulkan is 64bit. Is the problem that it says (below) `nfd Platform: PLATFORM_WIN32`?
Any ideas on how to fault-trace are welcome. I just wanted to test this program, is there a compiled version available?
```
Dick@Koffein MINGW64 ~
$ cd ~/scopehal-apps/msys2

Dick@Koffein MINGW64 ~/scopehal-apps/msys2
$ MINGW_ARCH=mingw64 makepkg-mingw --noconfirm --noprogressbar -sCLf
==>  MINGW_ARCH: mingw64
  -> Building mingw64...
==> Making package: mingw-w64-scopehal-apps 0.0.0.r2212.gd48c741-1 (Fri Jan  6 03:06:08 2023)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
==> Removing existing $srcdir/ directory...
==> Extracting sources...
==> Starting pkgver()...
==> Removing existing $pkgdir/ directory...
==> Starting build()...
CMake Warning at cmake/FindVulkan.cmake:523 (message):
  Library glslang for Debug configuration is missing, imported target
  Vulkan::glslang may not be able to link when targeting this build
  configuration due to incompatible MSVC ABI.  Consider re-installing the
  Vulkan SDK and request debug libraries to fix this warning.
Call Stack (most recent call first):
  cmake/FindVulkan.cmake:537 (_Vulkan_set_library_component_found)
  CMakeLists.txt:60 (include)

CMake Warning at cmake/FindVulkan.cmake:523 (message):
  Library shaderc_combined for Debug configuration is missing, imported
  target Vulkan::shaderc_combined may not be able to link when targeting this
  build configuration due to incompatible MSVC ABI.  Consider re-installing
  the Vulkan SDK and request debug libraries to fix this warning.
Call Stack (most recent call first):
  cmake/FindVulkan.cmake:544 (_Vulkan_set_library_component_found)
  CMakeLists.txt:60 (include)

-- Found Vulkan: C:/VulkanSDK/1.3.224.1/Lib/vulkan-1.lib (found version "1.3.224") found components: glslang shaderc_combined glslc glslangValidator
CMake Warning at cmake/FindVulkan.cmake:523 (message):
  Library glslang for Debug configuration is missing, imported target
  Vulkan::glslang may not be able to link when targeting this build
  configuration due to incompatible MSVC ABI.  Consider re-installing the
  Vulkan SDK and request debug libraries to fix this warning.
Call Stack (most recent call first):
  cmake/FindVulkan.cmake:537 (_Vulkan_set_library_component_found)
  src/glscopeclient/CMakeLists.txt:11 (find_package)

CMake Warning at cmake/FindVulkan.cmake:523 (message):
  Library shaderc_combined for Debug configuration is missing, imported
  target Vulkan::shaderc_combined may not be able to link when targeting this
  build configuration due to incompatible MSVC ABI.  Consider re-installing
  the Vulkan SDK and request debug libraries to fix this warning.
Call Stack (most recent call first):
  cmake/FindVulkan.cmake:544 (_Vulkan_set_library_component_found)
  src/glscopeclient/CMakeLists.txt:11 (find_package)

-- Found Vulkan: C:/VulkanSDK/1.3.224.1/Lib/vulkan-1.lib (found version "1.3.224") found components: glslc glslangValidator
nfd Platform: PLATFORM_WIN32
nfd Compiler: COMPILER_GNU
-- Configuring done
-- Generating done
-- Build files have been written to: C:/msys64/home/Dick/scopehal-apps/build
[1/15] cmd.exe /C "cd /D C:\msys64\home\Dick\scopehal-apps\build\lib\scopeprotocols\shaders && C:\msys64\mingw64\bin\cmake.exe -E true"
[2/15] cmd.exe /C "cd /D C:\msys64\home\Dick\scopehal-apps\build\src\glscopeclient\shaders && C:\msys64\mingw64\bin\cmake.exe -E true"
[3/15] cmd.exe /C "cd /D C:\msys64\home\Dick\scopehal-apps\build\src\ngscopeclient\shaders && C:\msys64\mingw64\bin\cmake.exe -E true"
[4/15] cmd.exe /C "cd /D C:\msys64\home\Dick\scopehal-apps\build\src\ngscopeclient\shaders && C:\msys64\mingw64\bin\cmake.exe -E true"
[5/15] Copying styles...
[6/15] Copying gradients...
[7/15] Copying shaders...
[8/15] Copying protocol shaders...
[9/15] Copying protocol shaders...
[10/15] Copying fonts...
[11/15] Copying icons...
[12/15] Copying icons...
[13/15] Creating dist/windows_x64...
[14/15] Creating Windows x64 MSI...
FAILED: src/glscopeclient/CMakeFiles/msi_windows_x64 C:/msys64/home/Dick/scopehal-apps/build/src/glscopeclient/CMakeFiles/msi_windows_x64
cmd.exe /C "cd /D C:\msys64\home\Dick\scopehal-apps\build && C:\msys64\mingw64\bin\cmake.exe -E copy C:/msys64/home/Dick/scopehal-apps/src/glscopeclient/wix/LICENSE.rtf C:/msys64/home/Dick/scopehal-apps/build && "C:\Program Files (x86)\WiX Toolset v3.11\bin\heat" dir dist/windows_x64 -gg -ke -scom -sreg -srd -cg AllFiles -dr APPLICATIONFOLDER -sfrag -template fragment -out glscopeclient-files.wxs && "C:\Program Files (x86)\WiX Toolset v3.11\bin\candle" -nologo glscopeclient-files.wxs -arch x64 -out glscopeclient-files.wixobj && "C:\Program Files (x86)\WiX Toolset v3.11\bin\candle" -nologo glscopeclient.wxs -arch x64 -out glscopeclient.wixobj -ext WixUIExtension -ext WixUtilExtension && "C:\Program Files (x86)\WiX Toolset v3.11\bin\light" -nologo glscopeclient-files.wixobj glscopeclient.wixobj -b dist/windows_x64 -out dist/glscopeclient-d48c741-windows-x64.msi -ext WixUIExtension -ext WixUtilExtension"
The system cannot find the path specified.
[15/15] Creating portable zip from dist/windows_x64...
ninja: build stopped: subcommand failed.
==> ERROR: A failure occurred in build().
    Aborting...

Dick@Koffein MINGW64 ~/scopehal-apps/msys2
```

bvernoux, 2023-01-06:
You have not followed the steps from the document
https://github.com/glscopeclient/scopehal-docs/blob/5c8b04f2a015c926a1be034d86f11f547202335a/section-gettingstarted.tex#L190
Follow all the steps exactly like written

Teddyz, 2023-01-08:
I thought I did so, I used the pdf version of the document. 
Probably pebkac, skipping this.

closed 2023-01-08 by Teddyz

bvernoux, 2023-01-08:
You have probably missed to install WiX Toolset v3.11 in "C:\Program Files (x86)\WiX Toolset v3.11"