ngscopeclient/scopehal-apps #569: Windows CI builds don't install properly due to missing GTK libs

azonenberg, 2023-01-18:
We should look into bundling GTK redistributables with our binaries, at least until we have ngscopeclient usable standalone.

bvernoux, 2023-05-27:
Workaround about that I'm using locally based on a MSYS2/MINGW64 shell script + python3 script to find and copy all dependencies in BUILD_NAME directory (to be changed in Install_scopehal-apps_build.sh)

[Install_scopehal-apps_build_MSYS2_MINGW64.zip](https://github.com/glscopeclient/scopehal-apps/files/11581240/Install_scopehal-apps_build_MSYS2_MINGW64.zip)

bvernoux, 2023-06-09:
Issue fixed by PR https://github.com/glscopeclient/scopehal-apps/pull/595

bvernoux, 2023-06-12:
Fixed with merge of PR https://github.com/glscopeclient/scopehal-apps/pull/595

closed 2023-06-18 by azonenberg