ngscopeclient/scopehal-apps #574: Windows build from source does not complete

andreasmatz, 2023-03-11:
Dear all,
I want to get glscopeclient up and running to use with a Siglent SDS5000X. My platform is Windows 10 (not really by choice, but most of my other measurement software is windows only). I ran into a build problem that I cannot solve.

First, I installed glscopeclient (binary) in MSYS2. It connected to the scope fine, but there was no waveform but only errors:
```
<== [31;1mERROR:←[0m ReadWaveformBlock for wavedesc 0 failed
```

Afterwards, I uninstalled it again and followed the steps in the build section. 

Building the dependencies went fine. Building scopehal-apps failed at this step:
6. Execute makepkg-mingw in subdir MSYS2:
//...
MINGW_ARCH = mingw64 makepkg - mingw -- noconfirm -- noprogressbar - sCLf

This build command itself fails on step 464/465:
```
[464/465] Creating Windows x64 MSI...
FAILED: src/glscopeclient/CMakeFiles/msi_windows_x64 C:/msys64/home/Admin/scopehal-apps/build/src/glscopeclient/CMakeFiles/msi_windows_x64
cmd.exe /C "cd /D C:\msys64\home\Admin\scopehal-apps\build && C:\msys64\mingw64\bin\cmake.exe -E copy C:/msys64/home/Admin/scopehal-apps/src/glscopeclient/wix/LICENSE.rtf C:/msys64/home/Admin/scopehal-apps/build && "C:\Program Files (x86)\WiX Toolset v3.11\bin\heat" dir dist/windows_x64 -gg -ke -scom -sreg -srd -cg AllFiles -dr APPLICATIONFOLDER -sfrag -template fragment -out glscopeclient-files.wxs && "C:\Program Files (x86)\WiX Toolset v3.11\bin\candle" -nologo glscopeclient-files.wxs -arch x64 -out glscopeclient-files.wixobj && "C:\Program Files (x86)\WiX Toolset v3.11\bin\candle" -nologo glscopeclient.wxs -arch x64 -out glscopeclient.wixobj -ext WixUIExtension -ext WixUtilExtension && "C:\Program Files (x86)\WiX Toolset v3.11\bin\light" -nologo glscopeclient-files.wixobj glscopeclient.wixobj -b dist/windows_x64 -out dist/glscopeclient-5dff40f-windows-x64.msi -ext WixUIExtension -ext WixUtilExtension"
Das System kann den angegebenen Pfad nicht finden.
[465/465] Creating portable zip from dist/windows_x64...
ninja: build stopped: subcommand failed.
==> ERROR: A failure occurred in build().
    Aborting...
```

If we check the CMakeFiles folder, indeed the msi_windows_x64 is missing: 

MINGW64 ~/scopehal-apps/build/src/glscopeclient/CMakeFiles
# ls
glscopeclient.dir

I seem to be missing something, but I am not sure what went wrong. Do you have any ideas?
Best regards,
Andreas

Bueddl, 2023-03-12:
Hi!

one way to get started might be to look into what the CI does: https://github.com/glscopeclient/scopehal-apps/blob/master/.github/workflows/build-windows.yml (CI builds can be seen here https://github.com/glscopeclient/scopehal-apps/actions/workflows/build-windows.yml and you could also download the artifacts in case this helps)

Maybe you are missing some minor step here?

HTH
Sebastian

bvernoux, 2023-03-12:
It seems you have not installed WiX Toolset v3.11 in C:\Program Files (x86)\WiX Toolset v3.11\bin

andreasmatz, 2023-03-12:
Installing WiX Toolset fixed the issue. Thanks bvernoux! Closing this.

closed 2023-03-12 by andreasmatz