ngscopeclient/scopehal-apps #585: Linux build results in errors

KestutisMa, 2023-04-09:
Hi,
Nice app, thank you for your efforts on developing it.

What is recommended way to build glscope on Linux? 

I am using Arch Linux, tried install via AUR Linux https://aur.archlinux.org/packages/scopehal-apps-git 
but failed.
Also tried manually clone all git repos with dependencies. Cmake run was succesfull, but make failed with error:

```
[  0%] Building CXX object lib/scopehal/graphwidget/CMakeFiles/graphwidget.dir/Graph.cpp.o
[  0%] Linking CXX shared library libgraphwidget.so
[  0%] Built target graphwidget
[  0%] Building CXX object lib/scopehal/log/CMakeFiles/log.dir/log.cpp.o
[  0%] Building CXX object lib/scopehal/log/CMakeFiles/log.dir/ColoredSTDLogSink.cpp.o
[  0%] Building CXX object lib/scopehal/log/CMakeFiles/log.dir/STDLogSink.cpp.o
[  0%] Building CXX object lib/scopehal/log/CMakeFiles/log.dir/FILELogSink.cpp.o
[  1%] Linking CXX static library liblog.a
[  1%] Built target log
[  1%] Building CXX object lib/scopehal/xptools/CMakeFiles/xptools.dir/Socket.cpp.o
[  1%] Building CXX object lib/scopehal/xptools/CMakeFiles/xptools.dir/UART.cpp.o
[  2%] Linking CXX static library libxptools.a
[  2%] Built target xptools
[  2%] Building CXX object lib/scopehal/scopehal/CMakeFiles/scopehal.dir/cmake_pch.hxx.gch
In file included from /home/kest/glscope/from_github/scopehal-apps/build/lib/scopehal/scopehal/CMakeFiles/scopehal.dir/cmake_pch.hxx:6,
                 from <command-line>:
/usr/include/vulkan/vulkan_raii.hpp: In constructor ‘vk::raii::ShaderEXT::ShaderEXT(const vk::raii::Device&, const vk::ShaderCreateInfoEXT&, vk::Optional<const vk::AllocationCallbacks>)’:
/usr/include/vulkan/vulkan_raii.hpp:10192:9: error: ‘m_constructorSuccessCode’ was not declared in this scope
10192 |         m_constructorSuccessCode =
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [lib/scopehal/scopehal/CMakeFiles/scopehal.dir/build.make:77: lib/scopehal/scopehal/CMakeFiles/scopehal.dir/cmake_pch.hxx.gch] Error 1
make[1]: *** [CMakeFiles/Makefile2:1233: lib/scopehal/scopehal/CMakeFiles/scopehal.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

azonenberg, 2023-04-09:
That sounds like a problem with your Vulkan SDK installation.

Try installing a different SDK version, say 1.3.239 which is what I currently develop on? It's possible there's a regression in this version although an outright compile failure in the C++ wrapper layer would surprise me.

CraigLChapman, 2023-04-09:
Found this when investigating my own issues. It indeed looks to be a problem with the latest release (1.3.246) of Vulkan-Hpp, and an issue's already been raised under https://github.com/KhronosGroup/Vulkan-Hpp/issues/1557

I've reverted to 1.3.240 and all looks good.

azonenberg, 2023-04-09:
This is, incidentally, why our the build instructions in the manual call out a specific version of the SDK (currently 1.3.224 but I should update it; I now develop on .239). It's one we've tested on and know works; occasionally there are non-backward-compatible changes or outright bugs in the SDK that break things.

Keeping this ticket open until Khronos releases a fixed version of the SDK, but I don't think any action is needed on our side at the moment.

azonenberg, 2023-05-01:
Upstream Khronos issue is now fixed. 1.3.246 SDK is not compatible but the most recent release should be.

Closing as I don't think we need anything else on our end.

closed 2023-05-01 by azonenberg